pymodbus-dev/pymodbus #1204: Client async, timeout not handled correct for single calls.

janiversen, 2022-11-28:
The modbus requests waits in a asyncio.wait_for but it terminates at default timeout.

Mr-Wiredancer, 2023-01-21:
I have the same problem here. It doesn't show any useful error msg. Also if I use a sync client, it works ok.

janiversen, 2023-01-21:
It is hard to help when you not even tell which version you are using. If it is the same problem then the termination is in the log, and interesting to see. 

If there are nothing to see in the log how do you know you have a problem ?

janiversen, 2023-01-21:
You need to find the root cause, why your device is not responding….this issue is about how pymodbus handles that your device do not respond.

Mr-Wiredancer, 2023-01-23:
Sorry for not providing details. My intention was just to let you know that I encountered something similar. And at that moment I worked around by using sync version. Just got time to compile what happened. 

My code was following
```python
from pymodbus.client import AsyncModbusTcpClient
import asyncio

client = AsyncModbusTcpClient(IP, PORT)
asyncio.run(client.connect())
asyncio.run(client.read_holding_registers(61, 1))
```

and i have the following 
```shell
Traceback (most recent call last):
  File "/usr/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/jiahao/modbus-playground/env/lib/python3.10/site-packages/pymodbus/client/base.py", line 303, in execute
    resp = await asyncio.wait_for(req, timeout=self.params.timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```
I tried a timeout with 10s and  got the same output

the same configuration of synced version works:
```python
from pymodbus.client import ModbusTcpClient

client = AsyncModbusTcpClient(IP, PORT)
client.connect()
print(client.read_holding_registers(61, 1))
```

janiversen, 2023-01-23:
But the problem is still the same, your device do not respond within the timeout, so pymodbus raises a timeout exception which your add do not catch.

see our template call in examples/client_calls.py:
```
async def async_template_call(client):
    """Show complete modbus call, async version."""
    try:
        rr = await client.read_coils(1, 1, slave=SLAVE)
    except ModbusException as exc:
        txt = f"ERROR: exception in pymodbus {exc}"
        _logger.error(txt)
        raise exc
    if rr.isError():
        txt = "ERROR: pymodbus returned an error!"
        _logger.error(txt)
        raise ModbusException(txt)
    if isinstance(rr, ExceptionResponse):
        txt = "ERROR: received exception from device {rr}!"
        _logger.error(txt)
        # THIS IS NOT A PYTHON EXCEPTION, but a valid modbus message
        raise ModbusException(txt)

    # Validate data
    txt = f"### Template coils response: {str(rr.bits)}"
    _logger.debug(txt)
```

so in short, why does your device not respond within the timeout ?

Actually the sync version, do not really work with timeouts.

Mr-Wiredancer, 2023-01-24:
The device is a 3rd party one and the only thing I can do is to communicate with it using modbus, So I'm not sure what goes wrong there.

But my point is the sync version works and gets the correct response (within 3s), while the async one gets a timeout. Is it possible that the async version does something wrong?

alexrudd2, 2023-01-24:
I believe both of you are focusing on the hardware, but the problem is more likely to be in @Mr-Wiredancer 's understand of `asyncio`.  Async is written in a fundamentally different way than sync.

I respectfully recommend reading the Python `asyncio` docs regard "coroutines" and `asyncio.run()`
https://docs.python.org/3/library/asyncio-task.html#coroutines
https://docs.python.org/3/library/asyncio-runner.html#asyncio.run
>`asyncio.run` runs the passed coroutine, taking care of managing the asyncio event loop, finalizing asynchronous generators, and closing the threadpool.
> This function cannot be called when another asyncio event loop is running in the same thread.
[...]
> This function always creates a new event loop and closes it at the end. It should be used as a main entry point for asyncio programs, and should ideally only be called once.

What happens if you restructure your code to be more like the Python example?
```python
async def main():
    await asyncio.sleep(1)
    print('hello')

asyncio.run(main())
```

Mr-Wiredancer, 2023-01-24:
> I believe both of you are focusing on the hardware, but the problem is more likely to be in @Mr-Wiredancer 's understand of `asyncio`. Async is written in a fundamentally different way than sync.
>
> I [...]

Thanks for pointing it out. I read through the asyncio documentation and fixed the problem. It works perfect now.

janiversen, 2023-01-24:
@alexrudd2 you are right, I was too focused on the hardware, thanks for pointing it out.

closed 2023-01-24 by janiversen