pymodbus-dev/pymodbus #1209: Better Explanation of why to use strict = False on Windows machine

Conet1502, 2022-12-05:
@TheChatty if you are using this on a windows machine create the client with `strict=False` . That should fix the issue you are facing.

_Originally posted by @dhoomakethu in https://github.com/riptideio/pymodbus/issues/465#issuecomment-549203909_
      

The documentation for strict does not give a good explanation of this setting
https://pymodbus.readthedocs.io/en/latest/source/library/pymodbus.client.html

janiversen, 2022-12-05:
I think it is a correct documentation “ strict – (optional) Strict timing, 1.5 character between requests.” of course if you know the modbus protocol. Strict means there will always be a pause of 1.5 character (time depends on baudrate) between requests.

But pull requests are always welcome, if you have a better text.

Conet1502, 2022-12-05:
I understand the protocol, but related to #465, why would this matter on a Windows vs Linux platform?

janiversen, 2022-12-05:
#465 is based on very old code (2019), I do not think the statement is true today.

Conet1502, 2022-12-05:
Ok, just wondering if I was missing something.

janiversen, 2022-12-05:
and as you might have noticed it is not part of the documentation.

janiversen, 2022-12-05:
closing issue since it isn’t a issue

closed 2022-12-05 by janiversen