pymodbus-dev/pymodbus #1214: read_holding_registers fails (modbus rtu serial client)

tomatoma78, 2022-12-09:
### Versions

* Python: 3.10.2
* OS: windows 10 Pro 21H2
* Pymodbus: 3.0.2
* Modbus Hardware (if used): proprietary

### Pymodbus Specific
* Server: n/a
* Client: rtu - async

### Description

Serial connection to a reliable proprietary modbus device.
No communication problems using pymodbus 2.5.3
pymodbus 3.0.2: read_holding_registers sometimes fails to read the reply. It fails always if more than 2 registers are read.

### Code and Logs

```
# -*- coding: utf-8 -*-

from pymodbus.client import ModbusSerialClient as ModbusClient
import logging

def main():
    def setup_serial_client(port_name, port_baudrate, port_parity, port_timeout):
        # setup ModbusSerialClient
        # sets the number of stopbits according to the modbus protocol
        if port_parity == 'N':
            port_stopbits = 2
        else:
            port_stopbits = 1

        mb_client = ModbusClient(
            port     = port_name,
            baudrate = port_baudrate,
            parity   = port_parity,
            stopbits = port_stopbits,
            timeout  = port_timeout,
        )
        return mb_client

    port = 'COM3'
    timeout = 1
    baudrate = 19200
    parity = 'N'
    slave_id = 1

    logging.basicConfig()
    log = logging.getLogger()
    log.setLevel(logging.DEBUG)

    client = setup_serial_client(port, baudrate, parity, timeout)

    if client.connect():
        print(f'Connected to port {client.params.port}. baudrate: {client.params.baudrate}, '
              f'parity: {client.params.parity}, stopbits: {client.params.stopbits}, timeout: {client.params.timeout}')

        start_reg = 8195
        stop_reg = 8196
        print(f'Reading registers {start_reg}...{stop_reg}')
        registers = client.read_holding_registers(start_reg - 1, stop_reg - start_reg + 1, slave=slave_id)  # uint
        if registers.isError():
            print(f'Reading Error (registers {start_reg}...{stop_reg})\n')
        else:
            print(registers.registers)
            i = start_reg
            for reg in registers.registers:
                print(f'{i}, {reg}')
                i += 1
            print('')

        client.close()
        if client.is_socket_open():
            print("Serial port NOT closed!")
        else:
            print("Serial port closed!")

    else:
        print(f"Cannot open serial port {port}")

if __name__ == '__main__':
    main()
```

**Output OK:**

DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x20 0x2 0x0 0x2 0x6e 0xb
DEBUG:pymodbus.client.base:New Transaction state "SENDING"
DEBUG:pymodbus.transaction:Changing transaction state from "SENDING" to "WAITING FOR REPLY"
DEBUG:pymodbus.transaction:Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
DEBUG:pymodbus.transaction:RECV: 0x1 0x3 0x4 0x0 0x28 0x34 0x18 0x6c 0xf1
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0x4 0x0 0x28 0x34 0x18
DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse': 3]
Connected to port COM3. baudrate: 19200, parity: N, stopbits: 2, timeout: 1.0
Reading registers 8195...8196
[40, 13336]
8195, 40
8196, 13336

DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 1
DEBUG:pymodbus.framer.rtu_framer:Frame - [b'\x01\x03\x04\x00(4\x18l\xf1'] not ready
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
Serial port closed!

Process finished with exit code 0

**Output with error: (same source code)**

DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x20 0x2 0x0 0x2 0x6e 0xb
DEBUG:pymodbus.client.base:New Transaction state "SENDING"
DEBUG:pymodbus.transaction:Changing transaction state from "SENDING" to "WAITING FOR REPLY"
Connected to port COM3. baudrate: 19200, parity: N, stopbits: 2, timeout: 1.0
Reading registers 8195...8196
DEBUG:pymodbus.transaction:{msg_start} received, Expected 9 bytes Received 6 bytes !!!!
DEBUG:pymodbus.transaction:Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
DEBUG:pymodbus.transaction:RECV: 0x1 0x3 0x4 0x0 0x28 0x34
DEBUG:pymodbus.framer.rtu_framer:Frame - [b'\x01\x03\x04\x00(4'] not ready
DEBUG:pymodbus.transaction:Getting transaction 1
Reading Error (registers 8195...8196)

Serial port closed!
DEBUG:pymodbus.transaction:Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"

Process finished with exit code 0

janiversen, 2022-12-23:
I have now investigated the issue, and it seems to me that your device might delay the response from time to time:

DEBUG:pymodbus.transaction:{msg_start} received, Expected 9 bytes Received 6 bytes !!!!

Pymodbus only got 6 out of 9 bytes and thus fails. For serial connections this is typically a problem with the USB-RS485 converter or the rs485 wiring not being terminated correctly. 

You can try to use a higher timeout.

tomatoma78, 2022-12-23:
Hi Janiversen,

Thanks for looking at this issue.
My experience is that with pymodbus 2.5.3 the master can read up to 125 contiguous holding registers (the maximum allowed by the modbus protocol) without any error, while with pymodbus 3.0.2 the master cannot read more than a few contiguous holding registers, using the same HW. I have been using version 2.5.3 for many months (thousands of hours) with no problems. As soon as I installed pymodbus 3.0.2, I encountered this issue, so I rolled back to version 2.5.3.
I recorded the TX and RX signals with an oscilloscope and I couldn't see any problems.

Regards

janiversen, 2022-12-23:
If you looked with the oscilloscope, did you see 6 bytes or 9 bytes…it tells you if your device send the bytes correctly (which I believe), next level is your usb adapter, where you need to do the same read with another program.

I am regularly reading 96 registers (load curve) from my solar inverter without any problems. I use both serial and tcp.

If you only installed pymodbus 3.0.2 and did not modify your program that might also explain something, there are new parameters and some have gone away. 

The most likely problem though is that your timeout is too small, and does not allow sufficient time for the USB to pass on the message. The timeout handling did not work in 2.5.3, so that could explain it.

tomatoma78, 2022-12-27:
Hi Janiversen,

the slave reply is always correct, from what I can see from the oscilloscope. The Python code I used is the one I included in my first message, which was updated to pymodbus 3.0.2 (I hope I did it right). The timeout was set to 1 second (the unit is seconds, according to the documentation). The communication lasts 1 ms or so. I think I also tried with timeout = 10 seconds, but I faced the same problem.
If I am the only one having this problem, then there must be something wrong with my code or with my hardware. The HW works well with pymodbus 2.5.3, so, it looks like the HW is OK. The code is given above.
Or, perhaps, pymodbus 2.5.3 works well with my USB/RS-485 converter, while pymodbus 3.0.2 does not work well with my USB/RS-485 converter.

janiversen, 2022-12-27:
Version 2.5.3 and 3.0.2 uses the same pyserial library, so the USB is not likely to be the problem. You use the sync client, which have had a lot fewer changes than the async, so it is less likely (but still possible) to be a problem.

I note that you do not use the "close_comm" parameter or any of the other parameters we had added mostly to help with USB handling.

I cannot see any bigger problems in your code. The only think I can see is that pymodbus only receive 6 bytes but expected 9. That actually happens in you OK example as well as in the second example...so it seems your problem is more general.

Apart from that, something is off with your OK example:
'''
  DEBUG:pymodbus.framer.rtu_framer:Frame - [b'\x01\x03\x04\x00(4\x18l\xf1'] not ready
'''
I have no idea how "x00(4\x18l" come about, "(" and "l" are not something pymodbus writes.

tomatoma78, 2023-01-05:
Hi Janiversen,

using Pycharm, I created a new project from scratch with a new virtualenv and with my code included in my first post, with the following difference:
```
        stop_reg = 8197
```
... and a different COM port (COM7).

I installed in the virtualenv the following packages (via Pycharm Settings menu):
pip 22.3.1
pymodbus 3.0.2
pyserial 3.5
setuptools 65.6.3
wheel 0.38.4

Sometimes it works, sometimes it doesn't:
**Output OK:**

Connected to port COM7. baudrate: 19200, parity: N, stopbits: 2, timeout: 1.0
Reading registers 8195...8197
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x20 0x2 0x0 0x3 0xaf 0xcb
DEBUG:pymodbus.client.base:New Transaction state "SENDING"
DEBUG:pymodbus.transaction:Changing transaction state from "SENDING" to "WAITING FOR REPLY"
DEBUG:pymodbus.transaction:Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
DEBUG:pymodbus.transaction:RECV: 0x1 0x3 0x6 0x0 0x28 0x30 0x18 0x1 0x5d 0xe 0x1d
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0x6 0x0 0x28 0x30 0x18 0x1 0x5d
DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse': 3]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 1
DEBUG:pymodbus.framer.rtu_framer:Frame - [b'\x01\x03\x06\x00(0\x18\x01]\x0e\x1d'] not ready
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
[40, 12312, 349]
8195, 40
8196, 12312
8197, 349

Serial port closed!

Process finished with exit code 0

**Output NOT OK 1:**  (8 bytes received)

DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x20 0x2 0x0 0x3 0xaf 0xcb
DEBUG:pymodbus.client.base:New Transaction state "SENDING"
DEBUG:pymodbus.transaction:Changing transaction state from "SENDING" to "WAITING FOR REPLY"
Connected to port COM7. baudrate: 19200, parity: N, stopbits: 2, timeout: 1.0
Reading registers 8195...8197
DEBUG:pymodbus.transaction:{msg_start} received, Expected 11 bytes Received 8 bytes !!!!
DEBUG:pymodbus.transaction:Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
DEBUG:pymodbus.transaction:RECV: 0x1 0x3 0x6 0x0 0x28 0x30 0x18 0x1
DEBUG:pymodbus.framer.rtu_framer:Frame - [b'\x01\x03\x06\x00(0\x18\x01'] not ready
DEBUG:pymodbus.transaction:Getting transaction 1
Reading Error (registers 8195...8197)

Serial port closed!
DEBUG:pymodbus.transaction:Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"

Process finished with exit code 0

**Output NOT OK 2:**  (10 bytes received)

Connected to port COM7. baudrate: 19200, parity: N, stopbits: 2, timeout: 1.0
Reading registers 8195...8197
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x20 0x2 0x0 0x3 0xaf 0xcb
DEBUG:pymodbus.client.base:New Transaction state "SENDING"
DEBUG:pymodbus.transaction:Changing transaction state from "SENDING" to "WAITING FOR REPLY"
DEBUG:pymodbus.transaction:{msg_start} received, Expected 11 bytes Received 10 bytes !!!!
DEBUG:pymodbus.transaction:Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
DEBUG:pymodbus.transaction:RECV: 0x1 0x3 0x6 0x0 0x28 0x30 0x18 0x1 0x5d 0xe
DEBUG:pymodbus.framer.rtu_framer:Frame - [b'\x01\x03\x06\x00(0\x18\x01]\x0e'] not ready
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
Reading Error (registers 8195...8197)

Serial port closed!

Process finished with exit code 0

**Output NOT OK 3:** (9 bytes received)

Connected to port COM7. baudrate: 19200, parity: N, stopbits: 2, timeout: 1.0
Reading registers 8195...8197
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x20 0x2 0x0 0x3 0xaf 0xcb
DEBUG:pymodbus.client.base:New Transaction state "SENDING"
DEBUG:pymodbus.transaction:Changing transaction state from "SENDING" to "WAITING FOR REPLY"
DEBUG:pymodbus.transaction:{msg_start} received, Expected 11 bytes Received 9 bytes !!!!
DEBUG:pymodbus.transaction:Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
DEBUG:pymodbus.transaction:RECV: 0x1 0x3 0x6 0x0 0x28 0x30 0x18 0x1 0x5d
DEBUG:pymodbus.framer.rtu_framer:Frame - [b'\x01\x03\x06\x00(0\x18\x01]'] not ready
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
Reading Error (registers 8195...8197)

Serial port closed!

Process finished with exit code 0

The traces on the oscilloscope are always the same
CH1 (Top trace): slave receive
CH2 (Bottom trace): slave transmit

CH1: two overlapped examples
CH2 (white second from top): slave transmit example OK
CH2 (green, 3rd from top): slave transmit example not OK

[image]

The same traces with the two slave transmit aligned and overlapped:
[image]

The actual baudrate is correct (second cursor at the beginning of the last byte):
1/19200 * (1+8+2) * 10 = 5.729 ms
cursor time difference: 5.730 ms

The time from the last bit of slave receive to the start of slave transmit is 2.602 ms, which is more than 3.5 character times (1/19200 * 11 * 3.5 = approx 2 ms) as specified by the modbus RTU protocol.

The signal traces look always OK to me.
I repeat that I have no problems using pymodbus 2.5.3, while pymodbus 3.0.2 does not work with my setup.

I can't find where the problem is.

Thanks,
Regards,

janiversen, 2023-01-05:
Since pymodbus do not receive all bytes in the frame, and the slave does send them, the only weak link is the USB adapter you are using.

Looking at the commit log, I can see there are at least one commit, that might solve your problem:
https://github.com/riptideio/pymodbus/commit/94044f134014f90a1144809d6e93ab8a8606a8e5
This commit is available on dev, but not in v3.0.2 (will be in the upcoming v3.1.0). If you have the possibility please try with the current dev.

tomatoma78, 2023-01-05:
Hi Janiversen,

I created a new Pycharm project, updated pip, setuptools and wheel and installed pymodbus dev branch, with the following command in Pycharm terminal:

`pip install git+https://github.com/riptideio/pymodbus@dev
`
I can now read all contiguous holding registers (61) available in the slave I'm currently using.
**Problem solved.**

Thanks,
Best regards

janiversen, 2023-01-05:
super.

closed 2023-01-05 by janiversen