pymodbus-dev/pymodbus #1216: Modbus TCP server fails to process client's request

EagleOxy, 2022-12-10:
### Versions

* Python: Python 3.8.10
* OS: Ubuntu 20.04.5 LTS
* Pymodbus: 3.0.2

### Pymodbus Specific
* Server: tcp - sync
* Client: tcp - sync

### Description

Client and server implemented using pymodbus. Trying to read 50 holding registers from a Modbus TCP server results in "Exception Response(131, 3, SlaveFailure)"

### Code and Logs
Server
```python
import logging
from pymodbus.server import StartTcpServer
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext

logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

ip_address="localhost"
port=5020

def run_server():
	data_block = ModbusSlaveContext(hr=ModbusSequentialDataBlock(1, [17]*100)		)

	context = ModbusServerContext(slaves=data_block, single=True)

	StartTcpServer(contexts=context, address=(ip_address, port))

if __name__ == "__main__":
    run_server()
```
Server logs
```log
DEBUG:asyncio:Using selector: EpollSelector
DEBUG:pymodbus.server.async_io:Socket [('127.0.0.1', 5020)] opened
DEBUG:pymodbus.server.async_io:TCP client connection established [('127.0.0.1', 59176)]
DEBUG:pymodbus.server.async_io:Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x0 0x1 0x0 0x1
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x0 0x1 0x0 0x1
DEBUG:pymodbus.factory:Factory Request[ReadHoldingRegistersRequest': 3]
ERROR:pymodbus.server.async_io:Datastore unable to fulfill request: 'dict' object has no attribute 'validate'; Traceback (most recent call last):
  File "/home/orinux/.local/lib/python3.8/site-packages/pymodbus/server/async_io.py", line 271, in execute
    response = request.execute(context)
  File "/home/orinux/.local/lib/python3.8/site-packages/pymodbus/register_read_message.py", line 139, in execute
    if not context.validate(self.function_code, self.address, self.count):
AttributeError: 'dict' object has no attribute 'validate'

ERROR:pymodbus.pdu:Exception Response(131, 3, SlaveFailure)
DEBUG:pymodbus.server.async_io:send: [Exception Response(131, 3, SlaveFailure)]- b'000100000003008304'
DEBUG:pymodbus.server.async_io:Handler for stream [('127.0.0.1', 59176)] has been canceled
DEBUG:pymodbus.server.async_io:TCP client disconnected [('127.0.0.1', 59176)]
```
Client
```python
import logging
from pymodbus.client import ModbusTcpClient

logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

ip_address="localhost"
port=5020

client = ModbusTcpClient(ip_address, port)

client.connect()

result = client.read_holding_registers(1, 50)
print(result.registers)
```
Client logs
```log
DEBUG:pymodbus.client.tcp:Connection to Modbus server established. Socket ('127.0.0.1', 59176)
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x0 0x1 0x0 0x1
DEBUG:pymodbus.client.base:New Transaction state "SENDING"
DEBUG:pymodbus.transaction:Changing transaction state from "SENDING" to "WAITING FOR REPLY"
DEBUG:pymodbus.transaction:Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
DEBUG:pymodbus.transaction:RECV: 0x0 0x1 0x0 0x0 0x0 0x3 0x0 0x83 0x4
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x3 0x0 0x83 0x4
DEBUG:pymodbus.factory:Factory Response[131]
DEBUG:pymodbus.transaction:Adding transaction 1
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
Exception Response(131, 3, SlaveFailure)
```

janiversen, 2022-12-10:
This is a bit strange, without having looked deeply at it, I think it should work.

Seems the datastore might be referencing a wrong block, please try to define all 4 blocks (hr= …) with the same data_block, and see if that changes anything.

EagleOxy, 2022-12-10:
> This is a bit strange, without having looked deeply at it, I think it should work.
> 
> Seems the datastore might be referencing a wrong block, please try to define all 4 blocks (hr= …) with the same data_block, and see if that changes anything.

Hello Jan, I have tried the option of defining all the 4 blocks - the result is still the same.

janiversen, 2022-12-10:
try

datablock = ModbusSequentialDataBlock(0x00, [17] * 100)

and

context = ModbusSlaveContext(
                di=datablock, co=datablock, hr=datablock, ir=datablock, unit=1
            )

that should do the trix.

EagleOxy, 2022-12-11:
Modified the function code as per below, but no changes to the result.
```python
def run_server():
#	data_block = ModbusSlaveContext(
#		di=ModbusSequentialDataBlock(0, [17]*100),
#		co=ModbusSequentialDataBlock(0, [17]*100),
#		hr=ModbusSequentialDataBlock(0, [17]*100),
#		ir=ModbusSequentialDataBlock(0, [17]*100))

	data_block = ModbusSequentialDataBlock(0x00, [17]*100)
	store = ModbusSlaveContext(di=data_block, co=data_block, hr=data_block, ir=data_block, unit=1)
	context = ModbusServerContext(slaves=store, single=True)

	StartTcpServer(contexts=context, address=(ip_address, port))

janiversen, 2022-12-11:
Take a look at https://github.com/riptideio/pymodbus/blob/dev/examples/server_sync.py you can run that directly.

It uses a common setup function in: https://github.com/riptideio/pymodbus/blob/dev/examples/server_async.py

My suggested changes came from that example.

The example works ! Our test suite runs that automatically.

EagleOxy, 2022-12-11:
I am unable to run the examples above due to missing `examples` library.

janiversen, 2022-12-11:
???? you can simply clone the repository. Or download the 2 files and run them.

Anyhow you have to look for a difference between the working example and your code.

EagleOxy, 2022-12-12:
After cloning the repo and installing it via the commands:

> pip install -r requirements.txt

> pip install -e .

 I can confirm that the examples (`server_sync`.py & `server_async.py`) work - I am still unsure what the changes are between my server and the one from the examples, but considering I am a Python newbie, I must be doing something wrong.
 
 Something I observed is also that when I used the:

>  pip install -U pymodbus

It installs version 3.0.2, whereas the other method, installed the version 3.1.

janiversen, 2022-12-12:
Seems you did not read the README, it documents how to test releases.

do:
```
   git checkout v3.0.2
```

and you will have release 3.0.2, and then you will see that the examples work in your release.

You need to compare how you setup the datastore in your program to how it is setup in setup method I pointed you towards. The setup method have different ways, but find the call you use and compare.

EagleOxy, 2022-12-12:
Found the issue:
```python
StartTcpServer(contexts=context, address=(ip_address, port))
```

There's a typo in the StartTcpServer parameter. The fix:
```python
StartTcpServer(context=context, address=(ip_address, port))
```

janiversen, 2022-12-15:
Ok.

closed 2022-12-15 by janiversen

EagleOxy, 2022-12-15:
Thank you for your help!