pymodbus-dev/pymodbus #1236: computeCRC() returns unnatural data requiring clients to perform unnatural manipulations

mstovenour, 2023-01-01:
https://github.com/riptideio/pymodbus/blob/8b3af2ac2689475fc70a470bbb0bfe3c32ef6af7/pymodbus/utilities.py#L197

Why swap bytes and return an int only to make callers do unnatural things like this?
```struct.pack(">H", computeCRC(cmd))```

should just perform that pack using "<" before returning the data...  Then make appropriate changes to checkCRC() and two other callers.  That big endian pack looks like an error on the client side.

If anyone agrees this needs to be changed, I'll submit a pull request.

janiversen, 2023-01-01:
the modbus protocol uses big envían, so that is a must. You can discuss whether the return should be an int or bytes.

You do not need these struct.pack in order to use computeCRC you can use the int directly or use .to_bytes() if you want it as bytes.

dhoomakethu, 2023-01-02:
What is the use case where it is required to compute CRC and validate it. The client should take care of it unless you are building your own client and trying to hack around things.

mstovenour, 2023-01-02:
Yes, I am looking to add support for a motor that implements custom functions 0x41-0x46, 0x64, 0x65 over serial RTU.  I'm looking for a simple way to extend the library for custom device implementations.  

Serial uses little indian not big indian and _only_ serial uses CRC; therefore the only use of calculateCRC() is for the _little_ indian case.  If you look at the implementation of calculateCRC() it is "swapping bytes" to put the int into little indian format; a real Frankenstein.  The little indian nature is clear when I look at the implementation of calculateCRC() and the two places it is used:  [search?q=computeCRC](https://github.com/riptideio/pymodbus/search?q=computeCRC)

janiversen, 2023-01-02:
We already have support for custom requests/responses, and by using that you do not need to think about CRC (handled to RtuFramer). see https://pymodbus.readthedocs.io/en/latest/source/example/custom_message.html

Please read the modbus standard, big endian is used independent of the transport type.

mstovenour, 2023-01-03:
Speaking of specifications, I find it more collaborative to _show_ some one a specification reference vs. just saying "please read the ... standard".  To keep this discussion collaborative, I've annotated the Modbus RTU diagram from the Modbus over Serial Line specification.  You can find this on [page 13](https://modbus.org/docs/Modbus_over_serial_line_V1_02.pdf#page=13):
[image]

> The CRC field is appended to the message as the last field in the message. When this is done, the low–order byte of the field is
appended first, followed by the high–order byte. The CRC high–order byte is the last byte to be sent in the message.

I'm talking successfully, using this library, to a Regal Beloit Century motor over serial and can definitively prove it it requires the CRC in little endian for serial comms.  The Century motor is using little endian for all values.  It also turns out that many implementations use little endian ordering for other values even though the latest Modbus specification says to use big endian.  At least for serial, various pieces of hardware use different register endian ordering.  Some Modbus frameworks include options to control byte ordering.  There is a good discussion with some links here:  [Byte order for data stored in several registers](https://minimalmodbus.readthedocs.io/en/stable/modbusdetails.html#byte-order-for-data-stored-in-serveral-registers)

If you still doubt what I'm saying about the CRC then maybe you can explain why computeCRC() has [this line of code](https://github.com/riptideio/pymodbus/blob/6224f71bf238ce4d6837cb437829a4b325a1d57f/pymodbus/utilities.py#L197)?
```
swapped = ((crc << 8) & 0xFF00) | ((crc >> 8) & 0x00FF)
return swapped
```
That line of code was a real head scratcher given that it was returning an **int**.  If it was returning bytes or a bytestring I would understand.  It works but the intent is not obvious and it still requires encoding before appending to a message.

I think you clearly answered my question:  "If anyone agrees this is worth changing, ..."  Answer:  No

dhoomakethu, 2023-01-03:
@mstovenour to be fair, that line of code was written 14years ago by original authors of this library, may be it can be optimised or may be it works just as it is. What @janiversen is suggesting is that there are wrappers or helpers or example with which you can handle custom message which meets your requirements. We don't want to unintentionally break something which might otherwise was working fine. If you think this change can be for the good, please feel free to test locally with your setup and raise a PR. Thank you.

mstovenour, 2023-01-03:
Thank you for the pointer to the example for ModbusBaseClient.register().  I've performed some simple tests of raw packets using send(), recv(), calculateCRC(), and checkCRC(), to quickly prove out the hardware device functions.  Last night I wrote a test client for this device modeled after the example in custom_message.html.  That worked out well.  I'll now create a set of complete classes specific to this device.  I'll publish that as a standalone repository with pymodbus as an external dependency.

I've already seen some additional areas for making this code base more maintainable but for now I'll avoid offering any more refactoring pull requests as I study the code base.  A simple refactor like this is a good litmus test.

janiversen, 2023-01-03:
Pull requests are welcome, especially if they come with tests, so we can easily verify they work. Making small PRs with simple refactoring are not only a good litmus test, but also a good way to get into the code and for us to see how you work.

I have a refactoring of the client pending because currently create_protocol creates a new object which is confusing.

janiversen, 2023-01-06:
Closing this for now, as it is not an issue. Looking forward to see Pull requests.

closed 2023-01-06 by janiversen