pymodbus-dev/pymodbus #1242: Async TCP server swallows listen errors

jaskij, 2023-01-09:
### Versions

* Python: 3.10
* OS: Arch Linux, up to date
* Pymodbus: 3.0.2
* Modbus Hardware (if used): n/a

### Pymodbus Specific
* Server: tcp async

### Description

When running on desktop, I forgot to change the port. On Linux a regular user cannot listen on port <1024. The exception is silent, and only shows where I hit Ctrl+C to close the program.

### Code and Logs

Seems straightforward enough to reproduce, under Linux start an async TCP server listening on `127.0.0.1:502`. I can provide a minimal reproduction in a separate repository if needed.

```
^CTraceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/jaskij/projects/engine/fosfory/pymodbus_test/pymodbus_test/__main__.py", line 39, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 636, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.10/asyncio/base_events.py", line 603, in run_forever
    self._run_once()
  File "/usr/lib/python3.10/asyncio/base_events.py", line 1868, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.10/selectors.py", line 469, in select
    fd_event_list = self._selector.poll(timeout, max_ev)
KeyboardInterrupt
Task exception was never retrieved
future: <Task finished name='Task-4' coro=<ModbusTcpServer.serve_forever() done, defined at /home/jaskij/.cache/pypoetry/virtualenvs/pymodbus-test-fOaPahLU-py3.10/lib/python3.10/site-packages/pymodbus/server/async_io.py:545> exception=PermissionError(13, "error while attempting to bind on address ('127.0.0.1', 502): permission denied")>
Traceback (most recent call last):
  File "/home/jaskij/.cache/pypoetry/virtualenvs/pymodbus-test-fOaPahLU-py3.10/lib/python3.10/site-packages/pymodbus/server/async_io.py", line 548, in serve_forever
    self.server = await self.loop.create_server(
  File "/usr/lib/python3.10/asyncio/base_events.py", line 1513, in create_server
    raise OSError(err.errno, 'error while attempting '
PermissionError: [Errno 13] error while attempting to bind on address ('127.0.0.1', 502): permission denied
```

janiversen, 2023-01-09:
Seems like a bug, the server should return the exception.

janiversen, 2023-01-15:
Solved in #1256, available on dev.

closed 2023-01-15 by janiversen