pymodbus-dev/pymodbus #1259: Docker publish fails on dev.

janiversen, 2023-01-14:
I have tried to restart the job, but it continues to fail.

janiversen, 2023-01-14:
@dhoomakethu do you mind taking a look at this ?

dhoomakethu, 2023-01-14:
Will check later today. Thanks

janiversen, 2023-01-14:
just fyi, I just updated another PR, and docker publish worked there...no idea why

janiversen, 2023-01-15:
The problem is the key of the certificate.

And on top of that there are a lot of warnings about methods being outdated and will be removed soon. 

Have you used an old skeleton ? I searched for examples and ours seems very complicated compared to what I saw, but I am just observing, since I do not have experience with docker.

janiversen, 2023-01-15:
@1261 did not solve the problem.

janiversen, 2023-01-15:
My comment was wrong, I was looking at the wrong examples, sorry about that.

But we still have a key problem.

PR #1268 (2023-01-15): Solve docker problem.

fixes #1259
closed 2023-01-15 by janiversen