pymodbus-dev/pymodbus #1276: Allow setting values from arbitrary iterables and mappings

jaskij, 2023-01-17:
### Versions

* Python: 3.10
* OS: Linux
* Pymodbus: 3.0.2
* Modbus Hardware (if used): n/w

### Description

`ModbusSparseDataBlock.setValues()` using a one-dimensional `numpy.ndarray` with registers values. This does not work. Instead of using individual values to set register values, what I got was the whole array assigned to that value. This requires a workaround of calling `.tolist()` on the array every time I'm setting values, or using a list` in the first place.

Looking at the code of `ModbusSequentialDataBlock`, we see (docstrings omitted):

```python
    def setValues(self, address, values):
        if not isinstance(values, list):
            values = [values]
        start = address - self.address
        self.values[start : start + len(values)] = values
```

I believe the check should actually be `if not isinstance(values, collections.abc.Iterable)`. This would treat any iterable the way lists are now, allowing for a wider range of use cases.

For sparse data block, the case is similar, although there is an additional test `isinstance(values, dict)`, which should probably be `isinstance(values, collections.abc.Mappin)`.

janiversen, 2023-01-17:
Your analysis sounds correct, Pull requests are welcome, and will be reviewed positively.

We maintain sequential and sparse datastore, but our efforts are for the moment concentrated on the simulator datastore.

janiversen, 2024-06-12:
Closing as partly solved.