pymodbus-dev/pymodbus #1280: async serial server isn't explicitly started by StartAsyncSerialServer

alexrudd2, 2023-01-24:
Now my python modbus server isn't replying to the client talking to it on the serial port. It was working with 3.1.0, it fails with 3.1.1, and it's not because of the logging changes. I'll investigate...

Meanwhile, found a typo:
tcp.py line 213:
`            if self.params.host.startswith("unit:"):`

"unit" should be "unix", since it's about unix sockets...

_Originally posted by @peufeu2 in https://github.com/pymodbus-dev/pymodbus/issues/1279#issuecomment-1400424302_

alexrudd2, 2023-01-24:
The call to `start.()` was indeed removed in [`925660`](https://github.com/pymodbus-dev/pymodbus/pull/1256/files#diff-2b868d687ff9a33d19e875a6454786b499fb360e483cd5fa94bbd826198eb7deL1153).

janiversen, 2023-01-24:
hmmm seems like a real bug. Pull requests are welcome.

PR #1282 (2023-01-24): add missing server.start()

sorry for that, copy/paste error.

fixes #1280
closed 2023-01-24 by janiversen