pymodbus-dev/pymodbus #1284: Web simulator documentation and examples missing.

janiversen, 2023-01-24:
The new web simulator misses some features, before being are main simulator:
- ReST API
- Documentation
- trace of calls.

### Versions

* Python:
* OS:
* Pymodbus:
* Modbus Hardware (if used):

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

### Code and Logs

```python
# code and logs here.

# please use the following to format logs when posting them here
import pymodbus
pymodbus.pymodbus_apply_logging_config()
```

efdx, 2024-07-29:
Hello,

I am currently contemplating on possibilities for automating use of the client. The ReST-API would be a convenience for me, and I could possibly create a skeleton for the build_json_* functions based on their html equivalents. But what kind of state would you expect for it at this point? At best, they would be filled, but I don't think I have the know-how/time to set up full integration tests for whatever I would do, much less fully document it. 

So in short: if anything is done, what is the expected outcome?

janiversen, 2024-07-29:
Not stale

janiversen, 2024-07-29:
Documentation is of course important, without it the feature is unknown, however writing documentation is easier than programming skeletons, just have a look in docs

efdx, 2024-07-29:
> I am not sure how you can make a rest api without fully integrating it.

Yeah true, I guess it could not be called that :D What I meant is that if I take build_html_registers and convert on build_json_registers (more or less copy-paste + adapt at this point), I would get enough to manipulate registers with json format API. This indeed would not be a full API in any sense. Plus as said, I don't know enough of the features included in build_html_registers (mainly operations, Set, Clear, Stop, etc) to validate if they function properly or not. Due to these, I'd still call it a skeleton at best.

> Documentation is of course important

Of course. But in my case, the same skeleton principle would apply. It would only be something like

```text
ReST API Title
==============

Note. This is still a work in progress. Here is the current description of calls supported at the moment.

<insert-content-here>
```

But regardless. I'll see what I can do about the PR. No promises though.

janiversen, 2024-11-29:
Thinking about combining server and simulator so that the simulator will only be a http server using the server.