pymodbus-dev/pymodbus #1295: Server UnixSocket does not close cleanly (pytest show socket still open)

janiversen, 2023-01-25:
### Versions

* Python:
* OS:
* Pymodbus:
* Modbus Hardware (if used):

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

### Code and Logs

```python
# code and logs here.

# please use the following to format logs when posting them here
import pymodbus
pymodbus.pymodbus_apply_logging_config()
```

peufeu2, 2023-01-26:
```
    async def server_close(self):
        """Close server."""
        for k_item, v_item in self.active_connections.items():
            txt = f"aborting active session {k_item}"
            _logger.warning(txt)
            v_item.handler_task.cancel()
        self.active_connections = {}
```
```
        if self.server is not None:
            self.server.close()
            await self.server.wait_closed()
            self.server = None
```

I think these two blocks should be swapped so server is closed first and doesn't receive new connections after the active connections array has been wiped

janiversen, 2023-01-26:
test it submit a pull request.

janiversen, 2023-01-27:
Be aware that this problem is as far as I can tell only for the new Unix sockets, and since the are files the handling is a bit different.

janiversen, 2023-02-13:
solved.

closed 2023-02-13 by janiversen