pymodbus-dev/pymodbus #1297: ServerStop

HamishSwanbarton, 2023-01-25:
### Versions

* Python: 3.10
* OS: Windows 11
* Pymodbus: 3.1.1
* Modbus Hardware (if used): None

### Pymodbus Specific
* Server: tcp - sync
* Client: NA

### Description
I was trying to run multiple Modbus servers, and shut them down cleanly. However only one of the modbus servers gets shut down, the rest continue running. I expected the threads the servers were running in to stop, but all but one of them are still active.

### Code and Logs
This is my test that fails - an assertion error gets raised on the last line.
```python
import threading
import time
from pymodbus.server import StartTcpServer, ServerStop

import pymodbus

pymodbus.pymodbus_apply_logging_config()

n = 2
# Create n servers in threads
for i in range(n):
    thread2 = threading.Thread(target=StartTcpServer, kwargs={'context': None, 'address': ('127.0.0.1', 5020 + i)})
    thread2.start()

time.sleep(1)
assert threading.active_count() == n + 1
for _ in range(n):
    ServerStop()

time.sleep(1)
assert threading.active_count() == 1
```
Logs:
```
C:\Users\Hamish\Git\site-simulator\.venv310\Scripts\python.exe C:\Users\Hamish\Git\site-simulator\tests\wip.py 
Traceback (most recent call last):
  File "C:\Users\Hamish\Git\site-simulator\tests\wip.py", line 22, in <module>
    assert threading.active_count() == 1
AssertionError
```
I am able to get this test to pass, by adding a line to `ServerStop`, shown here:
```python
def ServerStop():  # pylint: disable=invalid-name
    """Terminate server."""
    if my_job := _serverList.get_server():
        if my_job.loop.is_running():
            asyncio.run_coroutine_threadsafe(my_job.async_await_stop(), my_job.loop)
            sleep(0.1)
            _serverList._remove(my_job)  # <---- This line causes test to pass
    else:
        raise RuntimeError("ServerStop called without server task active.")
```

janiversen, 2023-01-25:
You start the threads so you need to stop the threads, it is not in the hands of pymodbus.

Running multiple synchronous server is really asking for trouble. ServerStop() should always close the server on top of the list, what you have added should happen in my_job.async_await_stop(), but it seems to have been dropped.

Feel  free to make a pull request with your change.

janiversen, 2023-01-25:
I am actually impressed that it works at all, it was never intended to, nor tested to run multiple synchronous servers.

HamishSwanbarton, 2023-01-25:
Ah, thanks that is quite helpful. I shall restructure my code instead.

closed 2023-01-25 by HamishSwanbarton