pymodbus-dev/pymodbus #1300: Add constants for ModbusException

pkubanek, 2023-01-26:
Meaning for exception_code is somewhat standardized. Although of course, it depends on the manufacturer, there are Modbus guidelines listed there:

https://www.simplymodbus.ca/exceptions.htm

Can we have those codes defined as constants? Can we also add to the ExceptionResponse method to extract the string associated with the exception_code?

janiversen, 2023-01-26:
Sounds like a good idea, pull requests are welcome.

pkubanek, 2023-01-26:
This is already available with ModbusExcpetions, which I somehow missed:

```python
In [1]: import pymodbus.pdu

In [2]: pymodbus.pdu.ModbusExceptions.IllegalFunction
Out[2]: 1

In [3]: pymodbus.pdu.ModbusExceptions.decode(1)
Out[3]: 'IllegalFunction'

In [4]: pymodbus.pdu.ModbusExceptions.decode(5)
Out[4]: 'Acknowledge'
```

closed 2023-01-26 by pkubanek