pymodbus-dev/pymodbus #1319: pymodbus.server requires redis in 3.1.2

beenje, 2023-01-30:
### Versions

* Python: 3.11
* OS: Linux
* Pymodbus: 3.1.2
* Modbus Hardware (if used):

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

Running `pymodbus.server --help` raises an exception when `redis` isn't installed.

```
pymodbus.server --help
Traceback (most recent call last):
  File "/home/conda/feedstock_root/build_artifacts/pymodbus_1675064514324/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeho/bin/pymodbus.server", line 7, in <module>
    from pymodbus.repl.server.main import app
  File "/home/conda/feedstock_root/build_artifacts/pymodbus_1675064514324/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeho/lib/python3.11/site-packages/pymodbus/repl/server/main.py", line 16, in <module>
    from pymodbus.server.reactive.default_config import DEFAULT_CONFIG
  File "/home/conda/feedstock_root/build_artifacts/pymodbus_1675064514324/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeho/lib/python3.11/site-packages/pymodbus/server/__init__.py", line 5, in <module>
    from pymodbus.server.async_io import (
  File "/home/conda/feedstock_root/build_artifacts/pymodbus_1675064514324/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeho/lib/python3.11/site-packages/pymodbus/server/async_io.py", line 12, in <module>
    from pymodbus.datastore import ModbusServerContext
  File "/home/conda/feedstock_root/build_artifacts/pymodbus_1675064514324/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeho/lib/python3.11/site-packages/pymodbus/datastore/__init__.py", line 3, in <module>
    from pymodbus.datastore.database.redis_datastore import RedisSlaveContext
  File "/home/conda/feedstock_root/build_artifacts/pymodbus_1675064514324/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeho/lib/python3.11/site-packages/pymodbus/datastore/database/__init__.py", line 2, in <module>
    from pymodbus.datastore.database.redis_datastore import RedisSlaveContext
  File "/home/conda/feedstock_root/build_artifacts/pymodbus_1675064514324/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeho/lib/python3.11/site-packages/pymodbus/datastore/database/redis_datastore.py", line 5, in <module>
    import redis
ModuleNotFoundError: No module named 'redis'
```

See https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=649693&view=logs&jobId=656edd35-690f-5c53-9ba3-09c10d0bea97&j=656edd35-690f-5c53-9ba3-09c10d0bea97&t=e5c8ab1d-8ff9-5cae-b332-e15ae582ed2d

With 3.1.1, `redis` was an optional dependency.
I'd not expect this to change with a patch release.

Is this a bug that was introduced in 3.1.2?

Stausssi, 2023-01-30:
Same goes for python imports from `pymodbus.datastore`. Both `redis` and `sqlalchemy` are mandatory with 3.1.2

janiversen, 2023-01-30:
Ups my mistake, I will fix that fast.

janiversen, 2023-01-30:
Can you please test newest dev, I just added a fix and report back. If successful then I will make 3.1.3

Stausssi, 2023-01-30:
The import problems with `redis` and `sqlalchemy` are fixed. `pymodbus.server --help` now fails due to `typer` not being installed.

janiversen, 2023-01-30:
typer is an optional package (see requirements.txt line 28. pymodbus.server is part of the repl package.

janiversen, 2023-01-30:
Thanks for testing, I will make the .3 release today or tomorrow, I have another urgent problem to solve first.

PR #1320 (2023-01-30): Solve 1319

fixes #1319
closed 2023-02-02 by janiversen