pymodbus-dev/pymodbus #1322: PyModbus Errors in HASSIO

stefandbt, 2023-01-30:
### Versions

* Python: V3.10
* OS: HASSOS (Home Assistant)
* Pymodbus: V0.3.3.1
* Modbus Hardware (if used): Sungrow SG10RT

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

When trying to request values from my inverter via pymodbus (ModbusTCP2MQTT) everything was fine. Since last week, I am experiencing errors in the log and some values are not displayed:
 
### Code and Logs
```python
# code and logs here.

Getting these errors in log files:

2023-01-30 17:07:32.179 ERROR (SyncWorker_4) [pymodbus.factory] index out of range
2023-01-30 17:07:32.180 ERROR (SyncWorker_4) [pymodbus.transaction] Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")

# please use the following to format logs when posting them here
import pymodbus
pymodbus.pymodbus_apply_logging_config()
```

janiversen, 2023-01-30:
And just for reference new Homeassistant with newest hassos works nicely with pymodbus.

janiversen, 2023-02-02:
???

stefandbt, 2023-02-03:
Hi Jan,
sorry for the delay. In the meantime my Home Assistant installition had a HDD crash. I set it up again from scratch. I am using now ModbusTCP to connect to my SunGrow PV Powerplant. Working as expected at thew moment. Thanks for your help and sorry for the confusion.

Cheers, Stefan

closed 2023-02-03 by janiversen