pymodbus-dev/pymodbus #1330: Unwanted dependency on Redis and SQLAlchemy

scranen, 2023-02-04:
It seems that [this PR](https://github.com/pymodbus-dev/pymodbus/pull/1299), intended to improve documentation, had the side effect of making it impossible to import anything from the `datastore` package unless you have Redis and SQLAlchemy installed.

### Versions

* Python: any
* OS: any
* Pymodbus: 3.1.2
* Modbus Hardware (if used): n/a

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

I'm trying to import some classes from the datastore package to set up a mock. This worked before, but with the latest release of PyModbus this results in an error.

### Code and Logs

```python
from pymodbus.datastore import (
    ModbusSlaveContext, 
    ModbusServerContext,
    ModbusSparseDataBlock
)
```
Results in:
```
Traceback (most recent call last):
  File "/Users/scranen/Repos/fca/devices/test/mock_ptl_controller.py", line 6, in <module>
    from pymodbus.datastore import (
  File "/Users/scranen/.local/share/virtualenvs/devices-2hMeqm5U/lib/python3.10/site-packages/pymodbus/datastore/__init__.py", line 3, in <module>
    from pymodbus.datastore.database.redis_datastore import RedisSlaveContext
  File "/Users/scranen/.local/share/virtualenvs/devices-2hMeqm5U/lib/python3.10/site-packages/pymodbus/datastore/database/__init__.py", line 2, in <module>
    from pymodbus.datastore.database.redis_datastore import RedisSlaveContext
  File "/Users/scranen/.local/share/virtualenvs/devices-2hMeqm5U/lib/python3.10/site-packages/pymodbus/datastore/database/redis_datastore.py", line 5, in <module>
    import redis
ModuleNotFoundError: No module named 'redis'
```

janiversen, 2023-02-04:
Solved on dev and will be released as a patch fix very soon.

closed 2023-02-04 by janiversen

IgorVDubov, 2023-02-07:
the same thing in
/pymodbus/server/__init__.py
...
from pymodbus.server.simulator.http_server import ModbusSimulatorServer
breaks with import error if ModbusSimulatorServer not installed

IgorVDubov, 2023-02-08:
Sorry, my mistake,
when commenting