pymodbus-dev/pymodbus #1331: Parameter retry_on_empty and retries does not have any effect for serial client

jara1, 2023-02-04:
### Versions

* Python: 3.9.2
* OS: Debian
* Pymodbus: 3.1.2
* Modbus Hardware (if used): FTDI FT2232H

### Pymodbus Specific
 Client: rtu - sync

### Description
Parameter ```retry_on_empty``` and ```retries``` does not have any effect. Based on my debugging, it's not used from client context in transaction.py

As a result, there is no retry attempt.

### Code and Logs

```python
from pymodbus.client import ModbusSerialClient
import time
import logging
import pymodbus
pymodbus.pymodbus_apply_logging_config(logging.DEBUG)
client = ModbusSerialClient(method='rtu', port='/dev/ttyUSB1', baudrate=115200, timeout= 5, retry_on_empty=True, retries=10, debug=True)
client.connect()
while(1):
    rr = client.read_input_registers(address=0x3100, slave=1, count=1)
    print(rr)
    time.sleep(0.5)

client.close()

23:02:24 DEBUG logging:80 Running transaction 7
23:02:24 DEBUG logging:80 SEND: 0x1 0x4 0x31 0x0 0x0 0x1 0x3f 0x36
23:02:24 DEBUG logging:80 Changing state to IDLE - Last Frame End - 1675548144.052484 Current Time stamp - 1675548144.554936
23:02:24 DEBUG logging:80 New Transaction state "SENDING"
23:02:24 DEBUG logging:80 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
23:02:34 DEBUG logging:80 Transaction failed. (Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)) 
23:02:34 DEBUG logging:80 Frame - [b''] not ready
23:02:34 DEBUG logging:80 Getting transaction 1
23:02:34 DEBUG logging:80 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
23:02:35 DEBUG logging:80 Current transaction state - TRANSACTION_COMPLETE
```

janiversen, 2023-02-05:
hmm I thought that was fixed, let me take a look.

janiversen, 2023-03-03:
Corrected on dev.

closed 2023-03-03 by janiversen