pymodbus-dev/pymodbus #1333: Very slow performance of  client.write_register in TCP/IP

Tomberi211, 2023-02-09:
### Versions

* Python: 3.9.7
* OS: WIN 10

### Pymodbus Specific
* Client: Modbus TCP/IP

### Description

Hello everyone!

I am trying to control an inverter output power by using pymodbus. For that, I am trying to read some power values from there sources and make a very simple calculation to tell the inverter its output power.

In any case, I found that reading holding registers is quite quick and enough for my test, but **writing holding registers takes several seconds**. I have also read and tested that changing the timeout makes it quicker but it is also more unstable (as it closes communication a lot of times due to timeout).

I expect to have a similar performance by using client.write_register or by using client.read_register without adjusting the timeout. Can anybody give me a hand?

Here there is the full code, you can test it by using ModRSsim2 if you want.

PD: sorry for the Spanish-English mix

### Code and Logs

import time
from pymodbus.client import ModbusTcpClient

#Creamos un cliente Modbus TCP
client = ModbusTcpClient(host='localhost',port=502, timeout=1)
#Conectamos (inicializamos) el cliente
client.connect()

limitePotenciaAparenteGrid = 5 #KVAs

#Leer holding register
while True:
    lecturaPotenciaAparenteTotalInversores = client.read_holding_registers(0,1,1) #START,nRegisters, device ID
    lecturaPotenciaAparenteGrid = client.read_holding_registers(1,1,1) #START,nRegisters, device ID
    lecturaPotenciaAparenteLoad = client.read_holding_registers(2,1,1) #START,nRegisters, device ID
    
    
    # if success display registers
    if lecturaPotenciaAparenteTotalInversores:
        rAparenteInversor=lecturaPotenciaAparenteTotalInversores.registers
        # print('KVAs Inversores: ' % rAparenteInversor)
        print('KVAs Inversores: ' + str(rAparenteInversor[0]))
    else:
        print('unable to read KVAs Inversores')
        
    # if success display registers
    if lecturaPotenciaAparenteGrid:
        rAparenteGrid=lecturaPotenciaAparenteGrid.registers
        print('KVAs Grid: '+ str(rAparenteGrid[0]))
    else:
        print('unable to read KVAs Grid')
        
        
    # if success display registers
    if lecturaPotenciaAparenteLoad:
        rAparenteLoad=lecturaPotenciaAparenteLoad.registers
        print('KVAs Load: ' + str(rAparenteLoad))
    else:
        print('unable to read KVAs Load')    
    
    
    if rAparenteLoad[0] > limitePotenciaAparenteGrid:
        setPoint_KVAS_Inversores= rAparenteLoad[0]-limitePotenciaAparenteGrid
    else:
        setPoint_KVAS_Inversores=0;
        

    client.write_register(0,setPoint_KVAS_Inversores)
    calcGrid= rAparenteLoad[0]-setPoint_KVAS_Inversores
    client.write_register(1,calcGrid)

janiversen, 2023-02-09:
Hola

No problem with the spanish, I live in Granada.

The idea of timeout is the opposite of how you use it, it is for slow devices so they have time to reply. Do not try to set timeout to a very low value, as you have seem it leads to a lot of timeouts.

Your code looks OK, apart from that we normally add slave=. I have not time write_register for quite a while, but it should not take seconds !! 

Can you please add a debug log, so we can see where time is spent (it could be your device that is slow to respond). We test write_register in our test harness and there a write takes takes about 10 ms.

Tomberi211, 2023-02-09:
Hello janiversen,

Thank you for your quick answer! I have modified a little the code to have some time measurements, I don't know if this is enough information, I am quite new at python,  but if I can do anything more please let me know.

I have selected right know a timeout=4 seconds. It is very clear for me that, for some reason, it takes 4 seconds for each write, so 2*writte = 8 seconds

```

import time
from pymodbus.client import ModbusTcpClient

#Creamos un cliente Modbus TCP
client = ModbusTcpClient(host='localhost',port=502, timeout=4)
#Conectamos (inicializamos) el cliente
client.connect()

limitePotenciaAparenteGrid = 5 #KVAs

# do stuff

#Leer holding register
while True:
    
    t = time.time()
    
    lecturaPotenciaAparenteTotalInversores = client.read_holding_registers(0,1,1) #START,nRegisters, device ID
    lecturaPotenciaAparenteGrid = client.read_holding_registers(1,1,1) #START,nRegisters, device ID
    lecturaPotenciaAparenteLoad = client.read_holding_registers(2,1,1) #START,nRegisters, device ID
    
    
    # if success display registers
    if lecturaPotenciaAparenteTotalInversores:
        rAparenteInversor=lecturaPotenciaAparenteTotalInversores.registers
        # print('KVAs Inversores: ' % rAparenteInversor)
        print('KVAs Inversores: ' + str(rAparenteInversor[0]))
    else:
        print('unable to read KVAs Inversores')
        
    # if success display registers
    if lecturaPotenciaAparenteGrid:
        rAparenteGrid=lecturaPotenciaAparenteGrid.registers
        print('KVAs Grid: '+ str(rAparenteGrid[0]))
    else:
        print('unable to read KVAs Grid')
        
        
    # if success display registers
    if lecturaPotenciaAparenteLoad:
        rAparenteLoad=lecturaPotenciaAparenteLoad.registers
        print('KVAs Load: ' + str(rAparenteLoad))
    else:
        print('unable to read KVAs Load')    
    
    
    if rAparenteLoad[0] > limitePotenciaAparenteGrid:
        setPoint_KVAS_Inversores= rAparenteLoad[0]-limitePotenciaAparenteGrid
    else:
        setPoint_KVAS_Inversores=0;
     
    time2Read = time.time() - t    #Time to read 3 registers
    
    t = time.time()

    client.write_register(0,setPoint_KVAS_Inversores)
    calcGrid= rAparenteLoad[0]-setPoint_KVAS_Inversores
    client.write_register(1,calcGrid)
    
    time2Write = time.time() - t    #Time to write 2 registers
    
    print('Time used for reading: ' + str(time2Read))
    print('Time used for writing: ' + str(time2Write))
    print('')
```

Here are the results of the print: 

KVAs Inversores: 0
KVAs Grid: 0
KVAs Load: [0]
Time used for reading: 0.06352090835571289
Time used for writing: 8.03075098991394

If I know change the timeout to 2 seconds... :

`client = ModbusTcpClient(host='localhost',port=502, timeout=2)
`

KVAs Inversores: 0
KVAs Grid: 0
KVAs Load: [0]
Time used for reading: 0.0620880126953125
Time used for writing: 4.03085470199585

I don't know if the issue can be my own installation of the packages or what, but I tested this yesterday in the lab and it shows similar results (I did not exactly measure it as right now).

Thank you again for your help!!

Tomberi211, 2023-02-09:
> Hello janiversen,
>
> Thank you for your quick answer! I have modified a little the code to have some time measurements, I don't know if this is enough information, I am quite new at python, but if I [...]

> Hola
>
> No problem with the spanish, I live in Granada.
>
> The idea of timeout is the opposite of how you use it, it is for slow devices so they have time to reply. Do not try to set timeout to a very [...]

janiversen, 2023-02-09:
you need to add:
´´´
import logging

pymodbus.pymodbus_apply_logging_config(logging.DEBUG)
´´´
and upload the log, otherwise I cannot help you.

Tomberi211, 2023-02-09:
> you need to add: ´´´ import logging
> 
> pymodbus.pymodbus_apply_logging_config(logging.DEBUG) ´´´ and upload the log, otherwise I cannot help you.

Here is the log: 

KVAs Inversores: 10
KVAs Grid: 5
KVAs Load: [15]
13:29:27 DEBUG transaction:426 No response received to unbounded read !!!!
13:29:27 DEBUG transaction:428 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
13:29:27 DEBUG transaction:327 RECV: 
13:29:27 DEBUG socket_framer:172 Processing: 
13:29:27 DEBUG transaction:528 Getting transaction 19
13:29:27 DEBUG transaction:234 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
13:29:27 DEBUG tcp:247 Connection to Modbus server established. Socket ('127.0.0.1', 51150)
13:29:27 DEBUG transaction:130 Current transaction state - TRANSACTION_COMPLETE
13:29:27 DEBUG transaction:134 Running transaction 20
13:29:27 DEBUG transaction:291 SEND: 0x0 0x14 0x0 0x0 0x0 0x6 0x0 0x6 0x0 0x1 0x0 0x5
13:29:27 DEBUG base:210 New Transaction state "SENDING"
13:29:27 DEBUG transaction:312 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
13:29:29 DEBUG transaction:426 No response received to unbounded read !!!!
13:29:29 DEBUG transaction:428 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
13:29:29 DEBUG transaction:327 RECV: 
13:29:29 DEBUG socket_framer:172 Processing: 
13:29:29 DEBUG transaction:528 Getting transaction 20
13:29:29 DEBUG transaction:234 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
13:29:29 DEBUG tcp:247 Connection to Modbus server established. Socket ('127.0.0.1', 51151)
13:29:29 DEBUG transaction:130 Current transaction state - TRANSACTION_COMPLETE
13:29:29 DEBUG transaction:134 Running transaction 21
13:29:29 DEBUG transaction:291 SEND: 0x0 0x15 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x1
13:29:29 DEBUG base:210 New Transaction state "SENDING"
13:29:29 DEBUG transaction:312 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
13:29:29 DEBUG transaction:428 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
13:29:29 DEBUG transaction:327 RECV: 0x0 0x15 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0xa
13:29:29 DEBUG socket_framer:172 Processing: 0x0 0x15 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0xa
13:29:29 DEBUG factory:351 Factory Response[ReadHoldingRegistersResponse': 3]
13:29:29 DEBUG transaction:516 Adding transaction 21
13:29:29 DEBUG transaction:528 Getting transaction 21
13:29:29 DEBUG transaction:234 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
13:29:29 DEBUG transaction:130 Current transaction state - TRANSACTION_COMPLETE
13:29:29 DEBUG transaction:134 Running transaction 22
13:29:29 DEBUG transaction:291 SEND: 0x0 0x16 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x1 0x0 0x1
13:29:29 DEBUG base:210 New Transaction state "SENDING"
13:29:29 DEBUG transaction:312 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
13:29:29 DEBUG transaction:428 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
13:29:29 DEBUG transaction:327 RECV: 0x0 0x16 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x5
13:29:29 DEBUG socket_framer:172 Processing: 0x0 0x16 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x5
13:29:29 DEBUG factory:351 Factory Response[ReadHoldingRegistersResponse': 3]
13:29:29 DEBUG transaction:516 Adding transaction 22
13:29:29 DEBUG transaction:528 Getting transaction 22
13:29:29 DEBUG transaction:234 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
13:29:29 DEBUG transaction:130 Current transaction state - TRANSACTION_COMPLETE
13:29:29 DEBUG transaction:134 Running transaction 23
13:29:29 DEBUG transaction:291 SEND: 0x0 0x17 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x2 0x0 0x1
13:29:29 DEBUG base:210 New Transaction state "SENDING"
13:29:29 DEBUG transaction:312 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
13:29:29 DEBUG transaction:428 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
13:29:29 DEBUG transaction:327 RECV: 0x0 0x17 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0xf
13:29:29 DEBUG socket_framer:172 Processing: 0x0 0x17 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0xf
13:29:29 DEBUG factory:351 Factory Response[ReadHoldingRegistersResponse': 3]
13:29:29 DEBUG transaction:516 Adding transaction 23
13:29:29 DEBUG transaction:528 Getting transaction 23
13:29:29 DEBUG transaction:234 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
13:29:29 DEBUG transaction:130 Current transaction state - TRANSACTION_COMPLETE
13:29:29 DEBUG transaction:134 Running transaction 24
13:29:29 DEBUG transaction:291 SEND: 0x0 0x18 0x0 0x0 0x0 0x6 0x0 0x6 0x0 0x0 0x0 0xa
13:29:29 DEBUG base:210 New Transaction state "SENDING"
13:29:29 DEBUG transaction:312 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
Time used for reading: 0.06101560592651367
Time used for writing: 4.0458104610443115

janiversen, 2023-02-09:
Your device is pretty slow:
```
13:29:27 DEBUG transaction:291 SEND: 0x0 0x14 0x0 0x0 0x0 0x6 0x0 0x6 0x0 0x1 0x0 0x5
13:29:27 DEBUG base:210 New Transaction state "SENDING"
13:29:27 DEBUG transaction:312 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
13:29:29 DEBUG transaction:426 No response received to unbounded read !!!!
13:29:29 DEBUG transaction:428 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
13:29:29 DEBUG transaction:327 RECV:
```
Here you into a timeout after 2 seconds.

this looks normal:
```
13:29:29 DEBUG base:210 New Transaction state "SENDING"
13:29:29 DEBUG transaction:312 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
13:29:29 DEBUG transaction:428 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
13:29:29 DEBUG transaction:327 RECV: 0x0 0x16 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x5
```
Less than 1 second.

I cannot find your 4 seconds (apart from the reconnect, due to a short timeout) ? Setting timeout to something like 10 seconds, makes a lot more sense, since this is only to prevent we wait forever if the server decides to not respond.

Tomberi211, 2023-02-09:
Hello @janiversen ,

I understand what is the timeout for and I completely agree with you, but I find that the write waits until timeout to exit. Here another test with timeout= 10. The time used is now 20s:

14:22:56 DEBUG tcp:247 Connection to Modbus server established. Socket ('127.0.0.1', 53297)
14:22:56 DEBUG transaction:130 Current transaction state - IDLE
14:22:56 DEBUG transaction:134 Running transaction 1
14:22:56 DEBUG transaction:291 SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x1
14:22:56 DEBUG base:210 New Transaction state "SENDING"
14:22:56 DEBUG transaction:312 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
14:22:56 DEBUG transaction:428 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
14:22:56 DEBUG transaction:327 RECV: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0xa
14:22:56 DEBUG socket_framer:172 Processing: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0xa
14:22:56 DEBUG factory:351 Factory Response[ReadHoldingRegistersResponse': 3]
14:22:56 DEBUG transaction:516 Adding transaction 1
14:22:56 DEBUG transaction:528 Getting transaction 1
14:22:56 DEBUG transaction:234 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
14:22:56 DEBUG transaction:130 Current transaction state - TRANSACTION_COMPLETE
14:22:56 DEBUG transaction:134 Running transaction 2
14:22:56 DEBUG transaction:291 SEND: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x1 0x0 0x1
14:22:56 DEBUG base:210 New Transaction state "SENDING"
14:22:56 DEBUG transaction:312 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
14:22:56 DEBUG transaction:428 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
14:22:56 DEBUG transaction:327 RECV: 0x0 0x2 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x5
14:22:56 DEBUG socket_framer:172 Processing: 0x0 0x2 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x5
14:22:56 DEBUG factory:351 Factory Response[ReadHoldingRegistersResponse': 3]
14:22:56 DEBUG transaction:516 Adding transaction 2
14:22:56 DEBUG transaction:528 Getting transaction 2
14:22:56 DEBUG transaction:234 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
14:22:56 DEBUG transaction:130 Current transaction state - TRANSACTION_COMPLETE
14:22:56 DEBUG transaction:134 Running transaction 3
14:22:56 DEBUG transaction:291 SEND: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x2 0x0 0x1
14:22:56 DEBUG base:210 New Transaction state "SENDING"
14:22:56 DEBUG transaction:312 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
14:22:56 DEBUG transaction:428 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
14:22:56 DEBUG transaction:327 RECV: 0x0 0x3 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0xf
14:22:56 DEBUG socket_framer:172 Processing: 0x0 0x3 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0xf
14:22:56 DEBUG factory:351 Factory Response[ReadHoldingRegistersResponse': 3]
14:22:56 DEBUG transaction:516 Adding transaction 3
14:22:56 DEBUG transaction:528 Getting transaction 3
14:22:56 DEBUG transaction:234 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
14:22:56 DEBUG transaction:130 Current transaction state - TRANSACTION_COMPLETE
14:22:56 DEBUG transaction:134 Running transaction 4
14:22:56 DEBUG transaction:291 SEND: 0x0 0x4 0x0 0x0 0x0 0x6 0x0 0x6 0x0 0x0 0x0 0xa
14:22:56 DEBUG base:210 New Transaction state "SENDING"
14:22:56 DEBUG transaction:312 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
KVAs Inversores: 10
KVAs Grid: 5
KVAs Load: [15]
14:23:06 DEBUG transaction:337 Transaction failed. (Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)) 
14:23:06 DEBUG socket_framer:172 Processing: 
14:23:06 DEBUG transaction:528 Getting transaction 4
14:23:06 DEBUG transaction:234 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
14:23:06 DEBUG tcp:247 Connection to Modbus server established. Socket ('127.0.0.1', 53315)
14:23:06 DEBUG transaction:130 Current transaction state - TRANSACTION_COMPLETE
14:23:06 DEBUG transaction:134 Running transaction 5
14:23:06 DEBUG transaction:291 SEND: 0x0 0x5 0x0 0x0 0x0 0x6 0x0 0x6 0x0 0x1 0x0 0x5
14:23:06 DEBUG base:210 New Transaction state "SENDING"
14:23:06 DEBUG transaction:312 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
14:23:16 DEBUG transaction:426 No response received to unbounded read !!!!
14:23:16 DEBUG transaction:428 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
14:23:16 DEBUG transaction:327 RECV: 
14:23:16 DEBUG socket_framer:172 Processing: 
14:23:16 DEBUG transaction:528 Getting transaction 5
14:23:16 DEBUG transaction:234 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
14:23:16 DEBUG tcp:247 Connection to Modbus server established. Socket ('127.0.0.1', 53330)
14:23:16 DEBUG transaction:130 Current transaction state - TRANSACTION_COMPLETE
14:23:16 DEBUG transaction:134 Running transaction 6
14:23:16 DEBUG transaction:291 SEND: 0x0 0x6 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x1
14:23:16 DEBUG base:210 New Transaction state "SENDING"
[... 14 lines cut ...]
14:23:16 DEBUG transaction:327 RECV: 0x0 0x7 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x5
14:23:16 DEBUG socket_framer:172 Processing: 0x0 0x7 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x5
14:23:16 DEBUG factory:351 Factory Response[ReadHoldingRegistersResponse': 3]
14:23:16 DEBUG transaction:516 Adding transaction 7
14:23:16 DEBUG transaction:528 Getting transaction 7
[... cut ...]
14:23:16 DEBUG transaction:234 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
14:23:16 DEBUG transaction:130 Current transaction state - TRANSACTION_COMPLETE
14:23:16 DEBUG transaction:134 Running transaction 9
14:23:16 DEBUG transaction:291 SEND: 0x0 0x9 0x0 0x0 0x0 0x6 0x0 0x6 0x0 0x0 0x0 0xa
14:23:16 DEBUG base:210 New Transaction state "SENDING"
14:23:16 DEBUG transaction:312 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
Time used for reading: 0.048711538314819336
Time used for writting: 20.02159309387207

nt294, 2023-02-09:
I have been encountering this issue today, my write speeds were a lot longer than my read speeds, and it seemed to coincide with the timeout. (i.e., setting timeout to 2 secs results in 4 sec read, 5 secs is 10 sec read etc.) Turns out in my client.read calls I specified the slave ID, whereas in the write calls I didn't.

In my case, adding slave=0x1 as an argument to the write_registers method fixed the slow speed.

janiversen, 2023-02-09:
Sending a message, waits until a response is received OR timeout is reached. 

as an example:
´´´
14:23:16 DEBUG transaction:291 SEND: 0x0 0x7 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x1 0x0 0x1
14:23:16 DEBUG base:210 New Transaction state "SENDING"
14:23:16 DEBUG transaction:312 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
14:23:16 DEBUG transaction:428 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
14:23:16 DEBUG transaction:327 RECV: 0x0 0x7 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x5
14:23:16 DEBUG socket_framer:172 Processing: 0x0 0x7 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x5
14:23:16 DEBUG factory:351 Factory Response[ReadHoldingRegistersResponse': 3]
´´´
you see the whole cycle (sending request and receiving response) takes less than 1 second.

What happens is that in case of a no-response, the write waits for timeout seconds, then reconnects after another delay and retries, that is total takes a while.

In the log you send me, I cannot find anything with 4, 5 or 10 seconds cycle time. 

Please send me a log that show this long write time from the entry SENDING (which is when you call e.g. write_register) until “TRANSACTION_COMPLETE” which is when your call returns.

janiversen, 2023-02-13:
??

janiversen, 2023-02-18:
Closing issue as it seems solved.

closed 2023-02-18 by janiversen