pymodbus-dev/pymodbus #1337: Modbus Error: [Input/Output] [WinError 10054] An existing connection was forcibly closed by the remote host

Mininggamer, 2023-02-10:
Python: 3.10
pymodbus: 3.1.3 (I still meet the issus when using version 2.5.3 with ```pymodbus.client.sync.ModbusTcpClient``` )
I'm trying to write data to my PLC through modbus tcp/ip. Here is my code:

```python
from pymodbus.client import ModbusTcpClient

client = ModbusTcpClient("192.168.0.1", port = 502)
client.connect()

client.write_registers(0, 1)

```
It doesn't output anything. and my PLC doesn't receive any data.
If i use ```print(client.write_registers(0, 1)) ``` instead of ```client.write_registers(0, 1)``` the terminal show 

```
Modbus Error: [Input/Output] [WinError 10054] An existing connection was forcibly closed by the remote host
```

janiversen, 2023-02-11:
As it writes your device closes the connection. Are you sure your device allows writing to address 0 ?

janiversen, 2023-02-11:
Closing issue, since it is not a problem with the library.

closed 2023-02-11 by janiversen