pymodbus-dev/pymodbus #1338: Can't run pymodbus.simulator --help

beenje, 2023-02-13:
### Versions

* Python: 3.10.6
* OS: Linux
* Pymodbus: 3.1.3
* Modbus Hardware (if used):

### Description

Trying to run `pymodbus.simulator --help` fails:

```
<coroutine object main at 0x7efcc073cf90>
sys:1: RuntimeWarning: coroutine 'main' was never awaited
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
```

The `main` function used as entry point for the simulator is an async function: https://github.com/pymodbus-dev/pymodbus/blob/12859d0b82cc215a18ac757fe9319cdf1f9ec890/pymodbus/server/simulator/main.py#L113

It can't be used directly as an entry point. The entry point should be a function using `asyncio.run`.

janiversen, 2023-02-13:
Looks like a bug, but the workaround is simple.

closed 2023-02-13 by janiversen

beenje, 2023-02-14:
Thanks for the quick fix!

PR #1339 (2023-02-13): main() should be sync

fixes #1338