pymodbus-dev/pymodbus #1342: Issue when I run with python 2

georgethomas08, 2023-02-14:
Hi, 

I'm trying to poll the data from a modbus meter, it is working fine with python 3.8< but I have an embedded linux gateway which has only python2.7, I have downgraded the library version to 2.5.3 and installed pip2. still I could get error cleared 

please find the program below

**********************************************************************************************************************************************************

#!/usr/bin/env python
"""
Pymodbus Asynchronous Client Examples
--------------------------------------------------------------------------

The following is an example of how to use the asynchronous modbus
client implementation from pymodbus with asyncio.

The example is only valid on Python3.4 and above
"""
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
# --------------------------------------------------------------------------- #
# specify slave to query
# --------------------------------------------------------------------------- #
# The slave to query is specified in an optional parameter for each
# individual request. This can be done by specifying the `unit` parameter
# which defaults to `0x00`
# --------------------------------------------------------------------------- #

UNIT = 0x0A

client = ModbusClient(method='rtu', port='/dev/ttyUSB0')

client.connect()

# connect to device

# set/set information
#pp=client.read_holding_registers(address=0, count=114, slave=10)
rr = client.read_input_registers(0, 114, unit=UNIT)
data = rr.registers

print(data)
print()
# print(meter_reg)

#for i in range(1,115):
#    print("reg_"+str(i)+":"+str(data[i],))
#print(data[34],)
# disconnect device
client.close()

*******************************************************************************************************************************************************

ERROR 
Traceback (most recent call last):
  File "modbus_read3.py", line 34, in <module>
    data = rr.registers
AttributeError: 'ModbusIOException' object has no attribute 'registers'

janiversen, 2023-02-14:
The error is because your device tells something is wrong, you need to run it again and make a debug log in order to see what happens.

Your program looks fine.

janiversen, 2023-02-14:
Please remember to downgrade pyserial as well.

dhoomakethu, 2023-02-14:
Check if you are not getting a error response with `rr.isError()` and then try to read registers.
```
if not rr.isError():
     data = rr.registers
```

janiversen, 2023-02-18:
Closing as a non-issue.

closed 2023-02-18 by janiversen