pymodbus-dev/pymodbus #1350: [question] 0 vs 1 indexing

jamesbraza, 2023-02-17:
### Versions

* Python: 3.10.9
* OS: macOS Monterey version 12.6
* Pymodbus: 3.1.1
* Modbus Hardware (if used): n/a

### Pymodbus Specific
* Server: tcp - sync
* Client: tcp - sync

### Description

My [Applied Motion MDX Servo Motor](https://www.applied-motion.com/products/integrated-servo-motors/mdxl62gn3?pn=MDXL62GN3IPB000) starts out its Modbus holding registers at 40001 in its [Host Command Reference](https://appliedmotion.s3.amazonaws.com/Host-Command-Reference_920-0002W_0.pdf):

[image]

I am using [WinTech](https://www.win-tech.com/html/demos.htm)'s ModScan64 version 6.A00-02.  It starts its holding registers at 40001:

[image]

In `pymodbus`, I found all of the registers are off-by-one (e.g. convert register 1 to 0), so its registers start at 40000:

```python
>>> from pymodbus.client import ModbusTcpClient
>>> client = ModbusTcpClient(MOTOR_IP_ADDRESS)
>>> assert client.connect()
>>> response = client.read_holding_registers(0, 8, slave=MODBUS_SLAVE_ID)
>>> response.registers
[0, 9, 31, 31, 0, 1265, 0, 1266]
```

My question: is `pymodbus` off by 1, or is AppliedMotion/WinTech wrong here?

Why do I need to decrement the manual's addresses by 1 to use `pymodbus`?

tracernz, 2023-02-17:
The Modbus protocol always used 0-based addresses, but the classic Modicon PLCs used 1-based addressing and so lots of implementations followed that. Modern Schneider (owns Modicon) PLCs now use 0-based addressing too, which makes it all fun when you have a mixed shop with multiple generations and multiple vendors. So the answer is, AppliedMotion are not wrong, just using the other convention.

janiversen, 2023-02-17:
@tracernz is totally correct, and that is why pymodbus have a parameter that changes the default addressing to be 1 based, just read the documentation.

And btw, having a mixed setup is not that difficult because you define pr device how you want to address it. 

Also the workaround is quite simple (but not needed) pymodbus uses 0 based addressing, so just add 1 to the address you need.

janiversen, 2023-02-17:
Closing as this is a non-issue, and covered by parameters.

closed 2023-02-17 by janiversen

jamesbraza, 2023-02-17:
Hi @janiversen and @tracernz, thanks for getting back!  Great answers and background.

> that is why pymodbus have a parameter that changes the default addressing to be 1 based

Searching in the docs, I see [`pymodbus.constants.Defaults.ZeroMode`](https://pymodbus.readthedocs.io/en/v3.1.3/source/library/constants.html#pymodbus.constants.Defaults.ZeroMode), but it seems `ZeroMode` only gets used in `ModbusSlaveContext` (which isn't relevant for my `ModbusTcpClient`).

Is there some way I can bias a particular `ModbusTcpClient` to have the addresses it reads/write be automatically incremented by 1, perhaps with a kwarg during instantiation?

janiversen, 2023-02-17:
Currently not, but pull requests are welcome. As I wrote the easy workaround is to increment the address you request by 1, read_holding(5) will read register 4.

We removed this parameter from the clients a while ago, because the workaround is just as simple.