pymodbus-dev/pymodbus #1354: Feature request: add hook to detect reconnect

wlcrs, 2023-02-18:
Both the RTU and TCP client have functionality that automatically do a reconnect when the connection is interrupted.

I would like to be notified when such an event takes place, so that I can take some appropriate actions.

There is no clear way to add this functionality in the client currently, and with your upcoming work in #1199 it is unwise for me to try to implement something myself.

Proposal: add an argument to the constructor which allows you to pass a function that will be called when the event takes place. Maybe it even makes sense to add callbacks for different events?

```
callbacks : {
 'on_connection_made': ...,
 'on_connection_lost': ....,
 'on_reconnect': ....
}
```
On `AsyncModbusXXXClient` objects these callback functions are async, otherwise they are expected to by normal functions?

janiversen, 2023-02-18:
Your idea of adding callbacks is good ! Please remember this needs to be added for async as well as sync and for all types.

I am not sure why you think there are no clear way, I would add the arguments to the base client (making it available for all) and then insert the hook in the appropriate methods.

The first part of cleaning the client is to clean the base, but I do not think it will affect a PR from you a lot, should it affect I can help solve the problems.

The callbacks should be defined as callbacks not async (this is the same as as “connection_made” etc). It is a lot more work to make them async, since you need to start an async task.

So please do not let #1199 stand in the way for a nice new functionality.

Feel free to make the pull request early in your process, and request comments from the maintainers.

janiversen, 2023-03-10:
On dev.

closed 2023-03-10 by janiversen