pymodbus-dev/pymodbus #1357: mypy reports invalid return value for execute() on ModbusClientMixin and its subclasses

alexrudd2, 2023-02-18:
```python
class ModbusClientMixin:  # pylint: disable=too-many-public-methods
    def execute(self, request: ModbusRequest) -> ModbusResponse:
        return request 

    def read_coils(
        self, address: int, count: int = 1, slave: int = 0, **kwargs: Any
    ) -> pdu_bit_read.ReadCoilsResponse:
        return self.execute(
            pdu_bit_read.ReadCoilsRequest(address, count, slave, **kwargs)
        )
```
```
pymodbus/client/mixin.py:54: error: Incompatible return value type (got "ModbusRequest", expected "ModbusResponse")  [return-value]
pymodbus/client/mixin.py:68: error: Incompatible return value type (got "ModbusResponse", expected "ReadCoilsResponse")  [return-value]
x20
```

@alexrudd2 proposed removing the type hint for the return value (`-> ModbusResponse`) from `ModbusClientMixin.execute()` and using `@abstractmethod`.  

> It only echoes request and does not return a ModbusResponse. All the subclasses which override execute return a subclass of ModbusResponse. Therefore, it looked like an abstract method, but I wasn't sure because of its use in pytest
_Originally posted by @alexrudd2 in https://github.com/pymodbus-dev/pymodbus/issues/1356#issuecomment-1435769793_

@janiversen disagreed:
> it returns a ModbusResponse when overwritten, so it should be defined as such. Remember this is an API spec for the library.
> This is an abstract method, but I strongly prefer not to define it as such, because it can be called from an app.
> we have a lot of methods that are overwritten but that does not make them abstract. only those that must be overwritten should be declared as abstract.
> The method can return a modbusResponse()
_Originally posted by @janiversen in https://github.com/pymodbus-dev/pymodbus/pull/1356#discussion_r1111114890_

alexrudd2, 2023-02-19:
`mypy` reports ~20 errors, and arguably they are valid.  
`ModbusClientMixin.execute()` returns `ModbusRequest`, which does not match the type hint of `ModbusResponse` (1 error)

A subclass such as `ModbusBaseClient.execute()` actually returns different class instances depending on the method. 
 Example: `ReadCoilsResponse` or `ReadDiscreteInputsResponse`.  The type hints for various methods that call `execute()` are these _specific classes_, which does not match the part class `ModbusResponse`. (~20 errors)

What I attempted:
(1) Do nothing, and silence the mypy errors => does not seem correct.

(2) Change the type hint for `ModbusClientMixin.execute()` to `ModbusRequest` to match what it actually returns.
=> this fixes the first error and the tests are OK.  But it seems strange that the overridden methods have a different signature, and doesn't fix the errors.
`pymodbus/client/base.py:180: error: Signature of "execute" incompatible with supertype "ModbusClientMixin"  [override]`

(3) Make `ModbusClientMixin.execute()` an abstract method / remove its return value.
=> This fixes the first error, but not the second.  Also, it breaks the tests.
```
    def test_client_mixin(arglist, method, arg, response):
        """Test mixin responses."""
        rr = getattr(ModbusClientMixin(), method)(**arglist[arg])
>       assert isinstance(rr, response)
E       AssertionError: assert False
E        +  where False = isinstance(None, <class 'pymodbus.register_read_message.ReadWriteMultipleRegistersRequest'>)
```

(4) Remove the type hint for `ModbusClientMixin.execute()`, but keep its return value.
=> This fixes all errors and the tests are OK.  However, @janiversen believes it ruins the type hints for apps which use the mixin.  (I'm confused who does this :) )

(5) Use `typing.TypeVar()` to create e.g. `ModbusResponseType` or `MR`.  Then change the type hints to use this instead of the current _class instance_.
```
MR = TypeVar('MR', bound=ModbusResponse)
class ModbusClientMixin:  # pylint: disable=too-many-public-methods
    def execute(self, request: ModbusRequest) -> MR:
        return request 

    def read_coils(
        self, address: int, count: int = 1, slave: int = 0, **kwargs: Any
    ) -> pdu_bit_read.ReadCoilsResponse:
        return self.execute(
            pdu_bit_read.ReadCoilsRequest(address, count, slave, **kwargs)
        )
```
```
pymodbus/client/mixin.py:45: error: A function returning TypeVar should receive at least one argument containing the same TypeVar  [type-var]
pymodbus/client/mixin.py:45: note: Consider using the upper bound "ModbusResponse" instead
```

(6) `typing.NewType()` ?
=> just mashes all the types together.

alexrudd2, 2023-02-19:
Who are the current users of `ModbusBaseClient.execute()`?  I only see the tests.  I cannot understand what possible use someone has for a method which simply returns the `Request` without doing anything.

janiversen, 2023-02-19:
Execute comments:
- The current users of execute are the apps using pymodbus ! The app use to handle non-implemented function codes.
- As I wrote earlier the best way to make it type proof, is to have execute return a ModbusResponse.
- Remark all the functions in mixin return either ModbusResponse or an inherited version (which probably is not done 100%)
- Using @abstractMethod in an API description is confusing for the users.
- Mixin is really an interface class, that are only used by the app.
- Mixin.execute are not used directly but with a transport implementation, but since mixin is an interface class @abstract should not be used (and it also have an implemtation, which @abstract should not have).

So what needs to be done is:
a) have mixin.execute return a ModbusResponse
b) secure that all the return classes are inherited from ModbusResponse

janiversen, 2023-02-19:
Here is a reference to how home assistant uses mypy.

https://github.com/home-assistant/core/blob/dev/mypy.ini

We should get at the same level (some day).

alexrudd2, 2023-02-27:
Closed in https://github.com/pymodbus-dev/pymodbus/pull/1377/commits/eb402a52f10aa4ad74b855f9b2249a643b18e87c

closed 2023-02-27 by alexrudd2