pymodbus-dev/pymodbus #1360: Set level for mypy.

janiversen, 2023-02-19:
### Versions

* Python:
* OS:
* Pymodbus:
* Modbus Hardware (if used):

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

### Code and Logs

```python
# code and logs here.

# please use the following to format logs when posting them here
import logging
import pymodbus
pymodbus.pymodbus_apply_logging_config(logging.DEBUG)
```

Once mypy is properly implemented (thanks to @alexrudd2), we should get to a level similar to:
https://github.com/home-assistant/core/blob/dev/mypy.ini

alexrudd2, 2023-02-27:
> Then I suggest (but you decide):
> - activate the options that are (relative) easy to solve
> - keep the rest deactivated
> - Group the ones you think we should activate for in a interface module
> - Group the rest as unwanted
> I have done a similar thing in the pylint setup, and it helps me remember that an option was tried and discarded etc.
> 
> You do not need to do it all in this PR, just put it up for review when you want.

_Originally posted by @janiversen in https://github.com/pymodbus-dev/pymodbus/issues/1392#issuecomment-1446612777_

janiversen, 2023-03-03:
I think we can safely close this now.

closed 2023-03-03 by janiversen