pymodbus-dev/pymodbus #1363: Should non-list values be allowed for multiple writes?

alexrudd2, 2023-02-19:
The type hints and tests for `write_registers()` in `ModbusClientMixin` indicate this method only accepts lists of values.

https://github.com/pymodbus-dev/pymodbus/blob/d2c7c778707d8e0cc1ea9ea30ffccf945ea56b89/pymodbus/client/mixin.py#L422-L442

However, for at least a decade (https://github.com/pymodbus-dev/pymodbus/commit/66bd765b9de81f181ba90d817a028928d6d3edb5) the underlying `WriteMultipleRegistersRequest` will actually accept single values and convert to a list.
https://github.com/pymodbus-dev/pymodbus/blob/d2c7c778707d8e0cc1ea9ea30ffccf945ea56b89/pymodbus/register_write_message.py#L149-L160

(1) Should this functionality be retained for `write_registers()`?  
If no, we're technically breaking backwards compatibility although it's unclear if anyone uses it.  `pymodbus` itself does not appear to.
If yes, we should 
- [ ] update docstrings
- [ ] Type hints become (`Union[int, float, str, List[Union[int, float, str]]]`)
- [ ] Add tests for this functionality

(2) Same for `readwrite_registers()`
(3) Same for `write_multiple_coils()`

alexrudd2, 2023-02-19:
Does Homeassistant use `write_registers(values=123)` or `write_registers(values=[123])`?

janiversen, 2023-02-20:
Just checked the code it actually uses  [123] so demanding a list() is ok.

alexrudd2, 2023-02-20:
Do you have access to GitHub advanced code search?  I have requested access to the beta but don't have it yet.  It would be best to search for other users of these functions.

Otherwise, I could `raise DeprecationWarning` for a while?

alexrudd2, 2023-02-20:
( I think) any other users which use `mypy` will already throw errors because the type signature is `List[Union[int, float, str]]`

Also, I recently learned that `Union[float, int]` is redundant: https://stackoverflow.com/questions/50928592/mypy-type-hint-unionfloat-int-is-there-a-number-type

janiversen, 2023-02-21:
No I do not have access, I prefer using google, since it gives a much broader perspective.

I am not sure where you would raise the warning, but it should not be in production code. You have lost me why that is needed. Just define it as list[int]

janiversen, 2023-02-21:
We do not accept float (that is more than 1 register and thus more than 1 value) nor do we allow str.

alexrudd2, 2023-02-21:
> No I do not have access, I prefer using google, since it gives a much broader perspective.
> I am not sure where you would raise the warning, but it should not be in production code. You have lost me why that is needed. Just define it as list[int]

`int` has been permitted for 10 years, and I am concerned there might be users that will break. Probably it's enough to list as an API change.

> We do not accept float (that is more than 1 register and thus more than 1 value) nor do we allow str.

OK, then the original typing in https://github.com/pymodbus-dev/pymodbus/pull/1056/ was too broad, and can be narrowed to `List[int]`

janiversen, 2023-02-21:
int have not been permitted for 10 years, I corrected an error in that function about 2 years ago :-)

Independent you forget that even though you add type hint, it is still only a hint, and not enforced, so it will break nothing for 99% of the users, and the 1% will understand they need to change.

Having written that have no problem accepting "int or list[int]", but float and str are simply wrong, and will not work with the current code (nor any future code).

alexrudd2, 2023-02-21:
> Independent you forget that even though you add type hint, it is still only a hint, and not enforced, so it will break nothing for 99% of the users, and the 1% will understand they need to change.

Sorry, I should have been explicit.  I intended to remove these lines that allow `int`.
https://github.com/pymodbus-dev/pymodbus/blob/5d7df3284615be7a4f16f2c16c5dc3e95b5e835d/pymodbus/register_write_message.py#L159-L160

janiversen, 2023-02-21:
If they do not cause you a mypy problem, then let's leave them, otherwise I am quite ok with removing them.

alexrudd2, 2023-02-21:
I did not search thoroughly, but found what appears to be at least one user of non-List values.  So instead I think it's better to change the type hints, docstrings, and tests.

https://github.com/binsentsu/home-assistant-solaredge-modbus/blob/4f11978e470490b85f65cceb30fb52874d4f7750/custom_components/solaredge_modbus/select.py#L123-L126

janiversen, 2023-02-21:
my my .... you came across HomeAssistant (although a custom component, with which I have had my challenges).

PR #1402 (2023-03-03): Add tests for writing multiple writes with a single value
Closes #1363

Currently there are only tests for e.g. `write_coils([list])`. This adds them for `write_coils(bool)`.

See also #1377
closed 2023-03-03 by janiversen