pymodbus-dev/pymodbus #1365: `RuntimeWarning: coroutine 'ModbusClientProtocol.execute' was never awaited`   in `AsyncModbusTcpClient.write_registers()`

Atokulus, 2023-02-20:
Dear PyModbus community
I hope you are all well 😄

Currently, I use the `AsyncModbusTcpClient` in a Python application.
When writing data on a register, the Python 3.11 interpreter is emitting a `RuntimeWarning` regarding that the 'ModbusClientProtocol.execute' was never awaited:

```sh
RuntimeWarning: coroutine 'ModbusClientProtocol.execute' was never awaited
```

This is due to the non-async method declarations in `mixin.py`.

My proposal is to duplicate and adapt `ModbusClientMixin` as a `AsyncModbusClientMixin`

Best regards
Markus

janiversen, 2023-02-20:
There are no need for 2 mixin modules, that is made to be used from both sync and async, however when using from async you need to (as normal) have an "await" in from of the function call.

```
   rr = await pymodbus.write...
```
As you can see in our examples that is how we do it.

janiversen, 2023-02-20:
Or read the documentation:

https://pymodbus.readthedocs.io/en/latest/source/library/client.html

Closing issue, feel free to reopen if this does not solve your problem.

closed 2023-02-20 by janiversen

Atokulus, 2023-02-20:
Thanks for @janiversen for the heads up 😅

Regarding type hinting, do you believe the following could be addressed in any way:

```sh
Class 'WriteMultipleRegistersResponse' does not define '__await__', so the 'await' operator cannot be used on its instances
```

janiversen, 2023-02-20:
We are working on adding type hinting so I assume there will be a solution. The warning is interesting though because the async method is "execute" which are defined in another class.