pymodbus-dev/pymodbus #1372: pip install -e does not work with setuptools >= 66.0.0

alexrudd2, 2023-02-21:
I am running setuptools 65.4.1, but only for pymodbus. I have venv for each project, and setuptools got installed, when I installed pymodbus. 

I actually wonder how setuptools got installed but I expect it is part of the python installation (which again is in the venv for pymodbus).

I did not know that it creates wheel, but I still think it is dangerous to suggest such installations to users. For a couple of reasons:
- your zip is a snapshot, not frozen in time, so we cannot checkout the version to reproduce problems (install, make a commit, install, and both have the same version).
- users that want to be on the bleeding edge should clone the repository, what allows them to get the latest commits easy.

apart from that, I noticed that "pip install -e ." also works with a standard python 3.11 installation.

I have however no problem in you recommending what you think is right,, but I do have a problem changing version on dev without a release. After a release we can (as I wrote earlier) decide if we want to change the scheme.

_Originally posted by @janiversen in https://github.com/pymodbus-dev/pymodbus/pull/1370#discussion_r1113333142_

janiversen, 2023-02-21:
The title is wrong, that depends a lot on the version of python and the platform you install it on. Furthermore this is not a pymodbus issue, currently pymodbus uses what is installed without making any demands on versions.

Setuptools are formally not installed with python, but in real life is normally part of the "vanilla" installation. I am running on a Mac mini M1, and both the vanilla python 3.10 and 3.11 works (pip install....).

We support 3.8+, and right now I do not know of any python standard installations < 3.11 that uses the new setuptools (it might very well exist though).

I am against changing version.py without making a release, that potentially breaks too many things. But when making a release we can surely change the scheme.

"Alpha" signals pre release, and are normally frozen with a tag, something we do not do on dev, so I am not sure that is the right way to identify a development version.

alexrudd2, 2023-02-21:
> The title is wrong, that depends a lot on the version of python and the platform you install it on. Furthermore this is not a pymodbus issue, currently pymodbus uses what is installed without making any demands on versions.

If you are so sure I am wrong, please run `python3 -m pip install --upgrade setuptools`  and try to install `dev` again.  I am convinced it will break with version 66, and have reproduced many times.

> Setuptools are formally not installed with python, but in real life is normally part of the "vanilla" installation. I am running on a Mac mini M1, and both the vanilla python 3.10 and 3.11 works (pip install....).

Python3.11 could not possibly include setuptools 66, since it was released earlier.  Your install will break at some point, either when you upgrade setuptools with pip or when a new Python release bundles a later version.

> We support 3.8+, and right now I do not know of any python standard installations < 3.11 that uses the new setuptools (it might very well exist though).

Who cares if it's distributed as a standard install?  Upgrading setuptools is perfectly acceptable on any install.

> "Alpha" signals pre release, and are normally frozen with a tag, something we do not do on dev, so I am not sure that is the right way to identify a development version.

OK, the only other alternatives I see are pinning `setuptools==65.0.0` permanently, or giving up `pip install`.  I don't care what the release names are, I just want to stop the breakage.

janiversen, 2023-02-21:
I never said you were wrong, I just said the vanilla installation do not have this problem ! I have not upgraded neither pip nor setuptools and do not intend to, I want to work with the vanilla version.

We test with standard python 3.8 - 3.11 so we actually do care what it is in the vanilla installations. I am sure at some point we need to do something, but that is a very generic statement, with every new python release there are adjustments.

Heh, you should start working on HomeAssistant, they have been using an old pip for years and are very reluctant to upgrade because it works. We cannot be responsible for what the user installs, we define and publish our test environments and that is what we support.

The responsibility is actually the reason why requirements.txt for most parts uses ==, we have had quite a number of requests to make that >=, but that would mean using untested versions.

I am not sure why you see this as such a big point, but please feel free to investigate the relationship between python version (vanilla installation) and setuptools version, if that can be expressed in requirements.txt, then that can be a solution.

Do however be aware, that it might complicate life for normal users e.g. on a Mac with python 3.10 that currently uses version 65.4.1.

To me it is really a non-issue, it seems the real issue is how we handle version.py, I have already said a couple of times, that I am open to changing the "x" after next release.

alexrudd2, 2023-02-21:
It seems we have a difference in philosophy.  I prefer to upgrade when possible to fix potential breakage early. 

does the attached PR work for you?  It prevents the breakage for people who upgrade, but does not cause any problems for "vanilla" users.

alexrudd2, 2023-02-21:
Also note 65.0.0 emits a warning, which does not appear to have any effect.
> WARNING: Built wheel for pymodbus is invalid: Metadata 1.2 mandates PEP 440 version, but '3.1.x' is not

closed 2023-02-21 by janiversen

janiversen, 2023-02-21:
??? Already merged. Setuptools 65.4.1 do not have a warning:
```
pip install -e .
Obtaining file:///Users/jan/repos/pymodbus
  Preparing metadata (setup.py) ... done
Requirement already satisfied: setuptools<66.0.0' in ./.venv/lib/python3.10/site-packages (from pymodbus===3.1.x) (65.4.1)
Installing collected packages: pymodbus
  Attempting uninstall: pymodbus
    Found existing installation: pymodbus 3.1.x
    Uninstalling pymodbus-3.1.x:
      Successfully uninstalled pymodbus-3.1.x
  Running setup.py develop for pymodbus
Successfully installed pymodbus-3.1.x

[notice] A new release of pip is available: 23.0 -> 23.0.1
[notice] To update, run: pip install --upgrade pip
(pymodbus) pymodbus % 
```

janiversen, 2023-02-21:
However there is another pip problem, which I do not know how to solve (it has to do with setup.cfg):
```
pip show pymodbus
Name: pymodbus
Version: 3.1.x
Summary: A fully featured modbus protocol stack in python
Home-page: https://github.com/pymodbus-dev/pymodbus/
Author: attr: pymodbus.__author__
Author-email: 
License: BSD-3-Clause
Location: /Users/jan/repos/pymodbus
Editable project location: /Users/jan/repos/pymodbus
Requires: setuptools
Required-by: 
(pymodbus) pymodbus % 
```
Thinking about I should open an issue.

alexrudd2, 2023-02-21:
> ??? Already merged. Setuptools 65.4.1 do not have a warning:
>
> ```
> pip install -e .
> Obtaining file:///Users/jan/repos/pymodbus
>   Preparing metadata (setup.py) ... done
> Requirement already satisfied: [...]

Possibly you have cached `pymodbus.egg-info`?  The warning is when a wheel is created, not installed.

pah, 2023-03-24:
Can this `setuptools<66.0` requirement be limited to development versions?

janiversen, 2023-03-24:
at the moment no. But pull requests are welcome.

alexrudd2, 2023-03-24:
>> Can this setuptools<66.0 requirement be limited to development versions?

> at the moment no. But pull requests are welcome.

Actually as of https://github.com/pymodbus-dev/pymodbus/commit/025ee8a4b0450e1d99e7d50d08c0eb4d08fb843f the pin can be removed.

However, a commit like https://github.com/pymodbus-dev/pymodbus/commit/6b27cd1e03dfe2391c8216a850250b3337b9c285 will break things again, since it uses 
`version = Version("pymodbus", 3, 2, "x", "")`.

If instead `dev` switches to something like this: 
`version = Version("pymodbus", 3, 2, 3, "alpha")` the problem is permanently avoided.

janiversen, 2023-03-25:
pull requests are welcome

janiversen, 2023-03-25:
another more interesting question, do we still need setup tools ? I am not sure if then how we use it.

alexrudd2, 2023-03-27:
> pull requests are welcome

It was previously rejected here: https://github.com/pymodbus-dev/pymodbus/pull/1370/commits/1ce4edeb9686fa7791206ce018d089fe54300d10

Since `3.2` has been released in the meantime, I created a new PR: https://github.com/pymodbus-dev/pymodbus/pull/1462

> another more interesting question, do we still need setup tools ? I am not sure if then how we use it.

From what I can understand, `setuptools` is needed to install the package at all.
https://github.com/pymodbus-dev/pymodbus/blob/533e0edc9c79fd3f80034c3799fafc0cbc0479be/setup.py#L8

janiversen, 2023-03-27:
If you carefully it was rejected at that time for a number of specific reasons relating to release issues.

I will take a look at your pr.

alexrudd2, 2023-03-27:
> If you carefully it was rejected at that time for a number of specific reasons relating to release issues.

Yes, I'm just including as context for other people (or even you & I in the future). :)

hroncok, 2023-04-21:
> From what I can understand, `setuptools` is needed to install the package at all.

I don't think setuptools are required on runtime, but the requirement is now specified as such (install_requires).

janiversen, 2023-04-21:
This issue is closed and have been solved long time ago. If you have a problem please open a new issue.

alexrudd2, 2023-04-21:
> This issue is closed and have been solved long time ago. If you have a problem please open a new issue.

No, he's right.  The latest released version `3.2.2` still has the pin, since https://github.com/pymodbus-dev/pymodbus/commit/c77e8586c1946f477507e7ebffd94f868f91525f is not part of that release.
https://github.com/pymodbus-dev/pymodbus/blob/a0481160cd392dace1096446d97c21dff6cff51a/requirements.txt#L15

It's been merged into `dev` and so the pin is will be removed in the next release.  
@hroncok is this causing a specific problem for you?

janiversen, 2023-04-21:
please do not continue discussions on a closed issue/PR. If the problem still exist on dev then please file a new issue.