pymodbus-dev/pymodbus #1374: pip show pymodbus, misses information.

janiversen, 2023-02-21:
```
pymodbus) pymodbus %  pip show pymodbus

Name: pymodbus
Version: 3.1.x
Summary: A fully featured modbus protocol stack in python
Home-page: https://github.com/pymodbus-dev/pymodbus/
Author: attr: pymodbus.__author__
Author-email: 
License: BSD-3-Clause
Location: /Users/jan/repos/pymodbus
Editable project location: /Users/jan/repos/pymodbus
Requires: setuptools
Required-by: 
```
Normally it gets the information from setup.cfg, but for some reason it does not work with "pip show".

alexrudd2, 2023-02-27:
`setuptools` apparently does not allow the use of `attr:`  for `author` or `maintainer`.

https://setuptools.pypa.io/en/latest/userguide/declarative_config.html#metadata.

I suggest moving them as strings directly into `setup.cfg`

janiversen, 2023-02-27:
Sounds like a good idea, thanks for investigating.

PR #1395 (2023-02-27): Correct setup.cfg.

fixes #1374
closed 2023-02-27 by janiversen