pymodbus-dev/pymodbus #1382: pymodbus.server does not start with its default parameters

chkr-private, 2023-02-21:
### Versions

* Python: 3.10.9
* OS: Fedora 36
* Pymodbus: 3.1.3

### Description

- starting "pymodbus.server run" fails, but starting it with "pymodbus.server run -s tcp" works ("-s tcp" should not be needed, it is the default)
- might be related to #1118 and #1139 
- it looks like that in https://github.com/pymodbus-dev/pymodbus/blob/dev/pymodbus/repl/server/main.py#L124 there is a mismatch between the expected type for that option ("str") and the used default parameter (the enum  ModbusServerTypes.tcp (which evaluates to "ModbusServerTypes.tcp" in a string context))

### Code and Logs
```
$ pymodbus.server run
20:21:10 ERROR logging:92 Invalid server ModbusServerTypes.tcp
$
$ pymodbus.server run -s tcp

__________                          .______.                    _________
\______   \___.__. _____   ____   __| _/\_ |__  __ __  ______  /   _____/ ______________  __ ___________
 |     ___<   |  |/     \ /  _ \ / __ |  | __ \|  |  \/  ___/  \_____  \_/ __ \_  __ \  \/ // __ \_  __ \\
 |    |    \___  |  Y Y  (  <_> ) /_/ |  | \_\ \  |  /\___ \   /        \  ___/|  | \/\   /\  ___/|  | \/
 |____|    / ____|__|_|  /\____/\____ |  |___  /____//____  > /_______  /\___  >__|    \_/  \___  >__|
           \/          \/            \/      \/           \/          \/     \/                 \/

SERVER >
```

janiversen, 2023-02-21:
This looks a lot like a bug, thanks for reporting it.

@dhoomakethu  can you please take a look ?

alexrudd2, 2023-02-21:
Actually, I noticed this yesterday but not investigate further. :(

closed 2023-02-21 by janiversen

dhoomakethu, 2023-02-21:
This was fixed ,looks like a regression on specific Python versions. Will take a look in detail this week.

PR #1384 (2023-02-21): Fix startup for REPL server
Closes #1382