pymodbus-dev/pymodbus #1385: pymodbus.simulator does not start

chkr-private, 2023-02-23:
### Versions

* Python: 3.10.9
* OS: Fedora 36
* Pymodbus: head of dev branch, commit 217469a234bc023a660acb9c448900288131022b

### Description
* start pymodbus.simulator, using the included setup.json file
* stops with error message
```
pymodbus.simulator --json_file pymodbus/server/simulator/setup.json 
22:17:54 INFO  logging:74 Start simulator
Traceback (most recent call last):
  File "/usr/bin/pymodbus.simulator", line 33, in <module>
    sys.exit(load_entry_point('pymodbus==3.1.3', 'console_scripts', 'pymodbus.simulator')())
  File "/home/chkr/projects/pymodbus/pymodbus/server/simulator/main.py", line 118, in main
    asyncio.run(run_main(), debug=True)
  File "/usr/lib64/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib64/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/chkr/projects/pymodbus/pymodbus/server/simulator/main.py", line 112, in run_main
    task = ModbusSimulatorServer(**cmd_args)
  File "/home/chkr/projects/pymodbus/pymodbus/server/simulator/http_server.py", line 142, in __init__
    device, custom_actions_dict or None
UnboundLocalError: local variable 'custom_actions_dict' referenced before assignment
```
* problem was introduced by commit 4d70ba559eeab598bffcdb241452a6aaaae30834 , PR #1369

### Comments
* after I reverted that commit, the simulator started
 * however, the web GUI did not show any registers and I was unable to retrieve the values of any register or coil (probably I misinterpreted setup.json) 
   * Should there be some data available with the default setup.json? Should the web GUI show some registers?
   * If yes, do you have an example how I can access them (e.g. via python.console)?

janiversen, 2023-02-24:
What a precise analysis, I am pleasantly surprised (NOT over the bug, but just how easy you made it to repair it).

The WEB GUI did show registers and allowed you to change values etc, but I have not tested it for a while. Btw. I am working on some extended documentation with web screen prints.

janiversen, 2023-02-24:
Just for info:
[image]

you select "registers" and then you need to define a filter...e.g. register 1-5, and press "add".

Defining multiple filters are done using "or" not "and".

I did however find another parameter bug, which is also solved in the PR.

janiversen, 2023-02-24:
If you come across people who want to help with CSS/HTML, then please send them here....the web pages needs someone better than me 🥇

PR #1386 (2023-02-24): Solve problem due to mypy changes

fixes #1385 

Interesting that the tests did not catch this, we need more tests to get coverage up around 100%.
closed 2023-02-24 by janiversen