pymodbus-dev/pymodbus #1393: [docs] Unused _thread_lock in performance module example

jamesbraza, 2023-02-27:
### Description

The below code is dead in the [performance module example](https://pymodbus.readthedocs.io/en/v3.1.3/source/examples.html#performance-module):

```python
_thread_lock = Lock()
```

I think this line should be either:
- Removed: if unused and irrelevant
- Inline commented: if unused but relevant
- Used to protect client access

### Additional Context

For what it's worth, this was also mentioned here: https://github.com/pymodbus-dev/pymodbus/discussions/1269#discussion-4759551

closed 2023-02-27 by janiversen

jamesbraza, 2023-02-27:
What happened is I searched the docs for `Thread`, and it's one of the few results.  I didn't realize that `2.5.3` examples are sort of just reference material at this point.

Just for posterity, https://github.com/pymodbus-dev/pymodbus/issues/1394 is related.

janiversen, 2023-02-27:
No problem, I named them 2.5.3, to indicate that they belong to that release.

Help in updating them is very appreciated, just submit a pull request.