pymodbus-dev/pymodbus #1394: [docs request] documenting thread safety

jamesbraza, 2023-02-27:
### Versions

* Python: 3.10.9
* OS: macOS Monterey and Windows 10+
* Pymodbus: 3.1.3
* Modbus Hardware (if used): n/a

### Pymodbus Specific

* Server: tcp - sync
* Client: tcp - sync

### Description

I am about to implement a `RLock` wrapper around my `ModbusTcpClient`, as I am pretty sure from [this discussion](https://github.com/pymodbus-dev/pymodbus/discussions/1269) that `pymodbus` is not thread safe.

I think it would be good to document that `pymodbus` is not thread safe, so that one can search the docs for "thread" and find this fact (and not GitHub discussions).

janiversen, 2023-02-27:
Sounds like a good idea, pull requests are welcome.

Be aware that instead of using multi threading, it is a lot more efficient to use asyncio and that is safe.

janiversen, 2023-03-03:
Actually pymodbus client is thread safe, provided you use different clients in each thread. 

Having multiple threads calling the same client is not supported, and will not be supported. That behaviour is only supported in the async client.

PR #1404 (2023-03-03): Add note about not being thread safe.

fixes #1394
closed 2023-03-03 by janiversen