pymodbus-dev/pymodbus #1398: AsyncModbusTcpClient raises exception when receive broken data which is uncatchable

salihgunay, 2023-02-28:
### Versions

* Python: 3.11
* OS: Macos and Linux
* Pymodbus: 3.1.3
* Modbus Hardware (if used):

### Pymodbus Specific
* Client: tcp - async

### Description

Hi,
I am using AsyncModbusTcpClient for tcp communication. I am using uvloop for event loop but this bug is not related to event loop. When I try to read_holding_register, if data is broken i get fatal error on transport TCPTransport error and I can not catch this exception for some reason. When I change the same code using ModbusTcpClient I get a normal ModbusIOException.  In order to reproduce this error I changed a simple echo server to always return "hello" as bytes.

### Code and Logs

```python
### Client Code Begin ###
from asyncio import set_event_loop
import uvloop

from pymodbus.client import AsyncModbusTcpClient

import logging
import pymodbus
pymodbus.pymodbus_apply_logging_config(logging.DEBUG)

async def hello():
    client = AsyncModbusTcpClient('localhost', port=4040, timeout=2)
    await client.connect()
    try:
        res = await client.read_holding_registers(1, 1, slave=1)
        logging.info(f"Result: {res}")
    except Exception as e:
        logging.error(f"Exception: {e}")

loop = uvloop.new_event_loop()
set_event_loop(loop)
loop.run_until_complete(hello())

### Client Code End ###

### Server Code Begin ###
import asyncio

class EchoServer(asyncio.Protocol):
    def connection_made(self, transport):
        self.transport = transport

    def data_received(self, data):
        print('data received: {}'.format(data))
        self.transport.write(b'hello')

loop = asyncio.get_event_loop()
coro = loop.create_server(EchoServer, '127.0.0.1', 4040)
server = loop.run_until_complete(coro)
print('serving on {}'.format(server.sockets[0].getsockname()))

try:
    loop.run_forever()
except KeyboardInterrupt:
    print("exit")
finally:
    server.close()

    loop.close()

### Server Code End ###

```

[image]

janiversen, 2023-02-28:
This is because we catch it internally. If you run it then we currently do not catch all exception. In both cases your call returns with the exception.

If you really want to, you can install your own decoder and catch it in there,

salihgunay, 2023-03-01:
Shouldn't It just give me ModbusIOException when receives invalid data just like synchronous ModbusTcpClient ? Because all I am changing is way of transportation.

janiversen, 2023-03-01:
It does, but as a return value, check what the function returns. Making real exceptions have caused quite a lot of problems, so we are removing those from the API.

janiversen, 2023-03-01:
And no, you are not changing the way of transportation, but the way the program works (sync or async).

salihgunay, 2023-03-01:
It doesn't return anything because of the exception which I can not catch as well. I had to do the following in order to catch the exception.

```python
from pymodbus.framer.socket_framer import ModbusSocketFramer

class MySocketFramer(ModbusSocketFramer):
    def _process(self, callback, error=False):
        try:
            super()._process(callback, error)
        except Exception as e:
            logging.error(e)

client = AsyncModbusTcpClient('localhost', port=4040, timeout=2, framer=MySocketFramer)

```

salihgunay, 2023-03-01:
Sorry for the exception. I get the TimeoutError. I didn't realize at first but I can catch the exception. But the ModbusIOException happens inside is handled by the event loop and it causes  a connection lost if I don't catch it

janiversen, 2023-03-01:
You are not supposed to catch the ModbusIOException, that is handled internally by pymodbus. It will close the connect and automatically reconnect.

salihgunay, 2023-03-01:
That's what I am trying to say. It doesn't catch for AsyncModbusTcpClient

janiversen, 2023-03-02:
I will have a look, we should catch that exception, close the connection and request a reconnect, which will cause a timeout in the application. There is a small thing that puzzles me in the log.

If we did not catch it, you should receive it in your “except Exception”.

salihgunay, 2023-03-02:
Exactly, I should get a ModbusIOException just like it does in synchronous client

janiversen, 2023-03-02:
Running on latest dev with:
```
async def _handle_holding_registers(client):
    """Read/write holding registers."""
    _logger.info("### write holding register and read holding registers")
    # _check_call(await client.write_register(2, 2, slave=SLAVE))
    rr = None
    try:
        rr = _check_call(await client.read_holding_registers(1, 1, slave=SLAVE))
    except Exception as exc:
        _logger.error(f"**GOT exception {type(exc)} - {exc}")
        exit(-1)
    assert rr.registers[0] == 10
```

you get:
´´´
19:40:23 ERROR client_test:98 **GOT exception <class 'pymodbus.exceptions.ConnectionException'> - Modbus Error: [Connection] Connection lost during request
´´´

Be aware that in a near future that will change, as we are working on removing exceptions on replace by an automatic reconnect.

closed 2023-03-02 by janiversen