pymodbus-dev/pymodbus #1415: mypy, acting different.

janiversen, 2023-03-09:
In pull request #1413 mypy is behaving differently in CI vs. ./check_ci.sh

./check_ci.sh do not report any problems, but mypy in the CI reports:
[image]

It would be nice if the checks could be identical.

janiversen, 2023-03-09:
Be aware that I corrected those 2 lines in the pr.

You can simply change List to list, in those 2 lines and test.

You can run CI on a branch.

alexrudd2, 2023-03-09:
OK, figured it out.

https://peps.python.org/pep-0585/

`list[int]` only works on Python3.9+.  `List[int]` works on all versions, including 3.8.

If you want `check_ci.sh` to function identically to the CI, it should be changed to `python3.8 -m mypy pymodbus`

janiversen, 2023-03-09:
Which would require having 3.8 installed, so lets not make that change.

I was actually looking to understand why, and you have explained it very nicely, thanks a lot.

closed 2023-03-09 by janiversen