pymodbus-dev/pymodbus #1419: pymodbus/payload.py - default byteorder and wordorder switched

Uncharger, 2023-03-09:
### Description

I am currently building a test client for a smart energy meter and looked into the code of the payload builder and decoder due to my client not working with default pymodbus settings.
There I found that for both the builder and decoder byteorder and wordorder are as follows:

` def __init__(
        self, payload=None, byteorder=Endian.Little, wordorder=Endian.Big, repack=False
    ):`
Lines 36 and 265.
Modbus specifications states that byteorder is always Big Endian ([doc](https://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf) 4.2 Data Encoding). While it is not stated explicitly about wordorder, it is common in the industry to use Little Endian, at least in my experience.

Is it possible they are accidentally switched, or is there a purpose I am missing for doing it this way in the lib?
Thank you for your answers.

janiversen, 2023-03-09:
For the last many years big endian have dominated the general computer market. If you by industry refer to SCADA system, there we have been using big endian for nearly all my working life (which happens to be quite long).

Modbus is BIG endian (which is actually the network protocol standard). Of course they do not say anything about word order, since they only consider registers (a word). But the general practical standard is to use big endian.

The payload class might have strange defaults, but I am not sure it really matters. Having said that pull requests are welcome.

If I remember right our client do not use the payload class, that is for the server part, we are moving away from using it slowly.

struct is you friend to test what is right and wrong on "big" / "little" endian and more importantly "network order". An alternative it to use int.to_bytes(test_value, 4, "big") vs "little" then you see the difference.

Uncharger, 2023-03-09:
Hello,
I did not explain myself well, badly even. The client is in C++, while I am using pymodbus to simulate a given amount of meters as slaves of a server. So as a test tool for the client.

Thank you for clarifying about the word order. I will will create a pull request regarding the default byteorder when I am not on the metro.
My curiosity is sated, thank you again.

janiversen, 2023-03-10:
But if you want new defaults, then feel free to submit a pull request.

janiversen, 2023-03-16:
Changing the defaults breaks existing code, and do add value to the library.

closed 2023-03-16 by janiversen