pymodbus-dev/pymodbus #1421: py.typed missing in pip install

MAKOMO, 2023-03-09:
### Versions

- Python: 3.11
- OS: macOS
- Pymodbus: 3.2.0 (pip install)
- Modbus Hardware (if used):

### Description

running mypy on my project complains about missing 

```
artisanlib/modbusport.py:68: error: Skipping analyzing "pymodbus.constants": module is installed, but missing library stubs or py.typed marker  [import]
artisanlib/modbusport.py:69: error: Skipping analyzing "pymodbus.payload": module is installed, but missing library stubs or py.typed marker  [import]
artisanlib/modbusport.py:241: error: Skipping analyzing "pymodbus.client": module is installed, but missing library stubs or py.typed marker  [import]
artisanlib/modbusport.py:385: error: Skipping analyzing "pymodbus.pdu": module is installed, but missing library stubs or py.typed marker  [import]
```

despite all your nice work on adding type annotations. The file py.typed id not get installed along via

```
# sudo -H python3 -m pip install pymodbus --upgrade
```

```
# cd /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pymodbus
# ls ./py.typed
ls: ./py.typed: No such file or directory
```

I added it as follows and the mypy errors went away.

```
# cd /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pymodbus
# sudo touch py.typed
```

janiversen, 2023-03-09:
@alexrudd2 this one is for you.

alexrudd2, 2023-03-09:
I did not notice since I use `python3 -m pip install -e .` Sorry!

PR #1422 (2023-03-10): Include py.typed in package
Closes #1421
closed 2023-03-10 by janiversen