pymodbus-dev/pymodbus #1434: StartAsyncSerialServer doesn't work pymodbus version 3.2???

dlmoffett, 2023-03-14:
### Discussed in https://github.com/pymodbus-dev/pymodbus/discussions/1433

<sup>Originally posted by **dlmoffett** March 13, 2023</sup>

### Versions

- Python: 3.9
- OS: Windows
- Pymodbus: 3.2.0
- Modbus Hardware (if used): USB to RS458

### Pymodbus Specific

- Server: `StartAsyncSerialServer`

### Description

`StartAsyncSerialServer` no longer actually starts the server.

### Code and Logs

```python
import asyncio
import logging

from pymodbus.framer.rtu_framer import ModbusRtuFramer
from pymodbus.server import StartAsyncSerialServer

logging.basicConfig(
    format="[%(asctime)s] [%(levelname)s] [%(name)s] %(message)s",
    datefmt="%Y-%m-%d %H:%M:%S %z",
    level=logging.INFO,
)
log = logging.getLogger(__name__)

async def modbus_slave():
    return await StartAsyncSerialServer(
        framer=ModbusRtuFramer,
        port="COM10",
        baudrate=19200,
    )

def sanity():
    log.info(f"✅ Here I am!")
    asyncio.run(modbus_slave())
    log.info(f"❌ I'm going insane!")
```

Per the [example documentation](https://pymodbus.readthedocs.io/en/dev/source/examples.html#asynchronous-server-example) I would expect to only see my first log statement and for the server to be up and running indefinitely until I send a keyboard interrupt. 

However, that's not the case and I see the following output when running the above code, which exits immediately after making the logs:
```
[2023-03-13 16:07:53 -0600] [INFO] [fix_pymodbus.sanity] ✅ Here I am!
[2023-03-13 16:07:53 -0600] [INFO] [fix_pymodbus.sanity] ❌ I'm going insane!
```

### Working Monkey Patch
If I make the following monkey patch everything works as expected:

```python
async def StartAsyncSerialServer(  # pylint: disable=invalid-name,dangerous-default-value
    context=None,
    identity=None,
    custom_functions=[],
    **kwargs,
):  # pragma: no cover
    """Start and run a serial modbus server.

    :param context: The ModbusServerContext datastore
    :param identity: An optional identify structure
    :param custom_functions: An optional list of custom function classes
        supported by server instance.
    :param kwargs: The rest
    """
    server = ModbusSerialServer(
        context, kwargs.pop("framer", ModbusAsciiFramer), identity=identity, **kwargs
    )
    await server.start()    # <----------------------Adding this makes it work 🤔
    await _serverList.run(server, custom_functions)
```

### Expected Behavior
I expect the sample code above to to run until a keyboard interrupt.

janiversen, 2023-03-14:
Since you already have the solution, feel free to submit a pull request (as I wrote earlier).

janiversen, 2023-03-14:
Seems like the correct place to me, and if not CI will tell.

PR #1443 (2023-03-19): add missing server.start().

fixes #1434
closed 2023-03-19 by janiversen