pymodbus-dev/pymodbus #1439: No longer able to specify unit type in read_holding_registers

ohshazbot, 2023-03-16:
### Versions

- Python: 3.9.14
- OS: Ubuntu 22.04LTS
- Pymodbus: 3.2.0
- Modbus Hardware (if used): N/A

### Pymodbus Specific

- Server: N/A
- Client: N/A

### Description

I'm upgrading some code from 2.5.3 to 3.x and bumped into this bug. Looks like when [units were added](https://github.com/pymodbus-dev/pymodbus/pull/1041), they were not prepared for units coming in as kwargs, resulting in 
```
Traceback (most recent call last):
  File "/home/john/workspace/collection/modbus/test.py", line 5, in <module>
    client.read_holding_registers(address=0, unit='VAH', count=1)
  File "/home/john/.pyenv/versions/3.9.14/envs/onboard-collection-modbus/lib/python3.9/site-packages/pymodbus/client/mixin.py", line 104, in read_holding_registers
    pdu_reg_read.ReadHoldingRegistersRequest(address, count, slave, **kwargs)
TypeError: __init__() got multiple values for argument 'unit'
```

read_holding_registers does not expose unit as an explicit argument so kwargs is the only option. Similar problems look like they exist in most of the functions in the client/mixin.py.

It's also totally possible that I'm doing something wrong, my Python isn't that great.

### Code and Logs

```python
from pymodbus.client import AsyncModbusTcpClient

client = AsyncModbusTcpClient(host='localhost', port=502)
client.connect()
client.read_holding_registers(address=0, unit='VAH', count=1)
```

janiversen, 2023-03-16:
unit was never meant to be something about how to interpret the values ( celcius, var etc...) it was the slave id and still is.

when you upgrade you will see that most unit= have been changed to slave=

janiversen, 2023-03-16:
If you used unit='VAN' you are lucky if it worked, it is probably interpreted as unit=0 meaning any slave.

closed 2023-03-16 by janiversen