pymodbus-dev/pymodbus #1447: Unable to decode response Modbus Error: Unknown response 0

austinmroczek, 2023-03-19:
### Versions

- Python: 3.10.10
- OS: Linux 5.15.49
- Pymodbus: 3.2.0
- Modbus Hardware (if used): unknown (reading from a NeoVolta solar battery system)

### Pymodbus Specific

- Server: tcp - async (not sure what is running on the battery, but it's talking to my TCP Async client)
- Client: tcp - async

### Description

I am building code to read values from a NeoVolta solar battery system.  I'm attempting to read values once per minute.  My code generally seems to work, but after some time (usually a few hours, but sometimes just a 5 or 10 minutes) I start to get ModbusIOException with "Unable to decode request", following "Unknown response 0".  I can see in the log that a response of the size I was expecting (100 registers) is returned.

### Code

```python

    self._client = AsyncModbusTcpClient(
            host=self._host,
            port=self._port,
            close_comm_on_error=True,
            reconnect_delay=1000,
            retry_on_empty=True,
    )

    self._get_value(0,100)

...

    async def _get_value(
        self,
        address: int,
        size: int,
        unit: int = 1,
        tries=1,
    ) -> any:

        if not self._client.connected:
            await self._client.connect()

        try:
            async with async_timeout.timeout(30):
                response = await self._client.read_holding_registers(
                    address=address, count=size, slave=unit
                )

```

### Logs

```txt

2023-03-19 22:24:24,371 DEBUG logging:104 Connecting.
2023-03-19 22:24:24.371 DEBUG (MainThread) [pymodbus.logging] Connecting.
2023-03-19 22:24:24,388 DEBUG logging:104 Client connected to modbus server
2023-03-19 22:24:24,389 INFO  logging:96 Protocol made connection.
2023-03-19 22:24:24.388 DEBUG (MainThread) [pymodbus.logging] Client connected to modbus server
2023-03-19 22:24:24.389 INFO (MainThread) [pymodbus.logging] Protocol made connection.
2023-03-19 22:24:24,390 INFO  logging:96 Connected to 192.168.1.205:8899.
2023-03-19 22:24:24.390 INFO (MainThread) [pymodbus.logging] Connected to 192.168.1.205:8899.
2023-03-19 22:25:23,133 DEBUG logging:104 send: 0x0 0x44 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x64
2023-03-19 22:25:23.133 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x44 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x64
2023-03-19 22:25:23,135 DEBUG logging:104 Adding transaction 68
2023-03-19 22:25:23.135 DEBUG (MainThread) [pymodbus.logging] Adding transaction 68
2023-03-19 22:25:23,574 DEBUG logging:104 recv: 0x0 0x44 0x0 0x0 0x0 0xcb 0x1 0x3 0xc8 0x0 0x3 0x0 0x1 0x2 0x1 0x32 0x31 0x30 0x38 0x32 0x39 0x39 0x32 0x36 0x38 0x0 0x0 0xff 0xff 0x0 0x0 0x7 0x17 0x0 0x0 0x52 0x14 0x13 0x4e 0x0 0x0 0x38 0x80 0x0 0x1 0x2 0x3 0x0 0x0 0x0 0xff 0x0 0xa 0x17 0x3 0x13 0xf 0x18 0x30 0x0 0x1 0x1f 0x40 0xff 0xff 0x0 0x0 0x0 0x0 0x12 0xc0 0x0 0x0 0x27 0x10 0xb 0xb8 0x1f 0x40 0x7 0xd0 0x3 0x52 0x0 0x0 0x0 0x0 0x3 0xe8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x21 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0xff 0xda 0x0 0x0 0x0 0x0 0xe2 0xfb 0xff 0xff 0x13 0xc4 0x5 0xd6 0xfd 0xe6 0x43 0x8a 0x0 0x0 0x0 0x30 0x0 0x1b 0x4c 0xfd 0x0 0x0 0x4d 0xbd 0x0 0x0 0x0 0x1 0x0 0x44 0x5 0x3b 0x17 0x70 0x0 0x0 0xc6 0x72 0x0 0x0 0x0 0x0 0x0 0x2c 0x7a 0xaa 0x0 0x0 0x1a 0x62 0x0 0x0 0x0 0x0 0x6 0x1e 0x5 0x94 0x0 0x0 0x0 0x0 0x7 0xd0 0x0 0x0 0x59 0x32 0x0 0x1 0xf7 0x7e 0xff 0xff
2023-03-19 22:25:23,575 DEBUG logging:104 Processing: 0x0 0x44 0x0 0x0 0x0 0xcb 0x1 0x3 0xc8 0x0 0x3 0x0 0x1 0x2 0x1 0x32 0x31 0x30 0x38 0x32 0x39 0x39 0x32 0x36 0x38 0x0 0x0 0xff 0xff 0x0 0x0 0x7 0x17 0x0 0x0 0x52 0x14 0x13 0x4e 0x0 0x0 0x38 0x80 0x0 0x1 0x2 0x3 0x0 0x0 0x0 0xff 0x0 0xa 0x17 0x3 0x13 0xf 0x18 0x30 0x0 0x1 0x1f 0x40 0xff 0xff 0x0 0x0 0x0 0x0 0x12 0xc0 0x0 0x0 0x27 0x10 0xb 0xb8 0x1f 0x40 0x7 0xd0 0x3 0x52 0x0 0x0 0x0 0x0 0x3 0xe8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x21 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0xff 0xda 0x0 0x0 0x0 0x0 0xe2 0xfb 0xff 0xff 0x13 0xc4 0x5 0xd6 0xfd 0xe6 0x43 0x8a 0x0 0x0 0x0 0x30 0x0 0x1b 0x4c 0xfd 0x0 0x0 0x4d 0xbd 0x0 0x0 0x0 0x1 0x0 0x44 0x5 0x3b 0x17 0x70 0x0 0x0 0xc6 0x72 0x0 0x0 0x0 0x0 0x0 0x2c 0x7a 0xaa 0x0 0x0 0x1a 0x62 0x0 0x0 0x0 0x0 0x6 0x1e 0x5 0x94 0x0 0x0 0x0 0x0 0x7 0xd0 0x0 0x0 0x59 0x32 0x0 0x1 0xf7 0x7e 0xff 0xff
2023-03-19 22:25:23,575 DEBUG logging:104 Factory Response[0]
2023-03-19 22:25:23,575 ERROR logging:120 Unable to decode response Modbus Error: Unknown response 0
2023-03-19 22:25:23.574 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x44 0x0 0x0 0x0 0xcb 0x1 0x3 0xc8 0x0 0x3 0x0 0x1 0x2 0x1 0x32 0x31 0x30 0x38 0x32 0x39 0x39 0x32 0x36 0x38 0x0 0x0 0xff 0xff 0x0 0x0 0x7 0x17 0x0 0x0 0x52 0x14 0x13 0x4e 0x0 0x0 0x38 0x80 0x0 0x1 0x2 0x3 0x0 0x0 0x0 0xff 0x0 0xa 0x17 0x3 0x13 0xf 0x18 0x30 0x0 0x1 0x1f 0x40 0xff 0xff 0x0 0x0 0x0 0x0 0x12 0xc0 0x0 0x0 0x27 0x10 0xb 0xb8 0x1f 0x40 0x7 0xd0 0x3 0x52 0x0 0x0 0x0 0x0 0x3 0xe8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x21 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0xff 0xda 0x0 0x0 0x0 0x0 0xe2 0xfb 0xff 0xff 0x13 0xc4 0x5 0xd6 0xfd 0xe6 0x43 0x8a 0x0 0x0 0x0 0x30 0x0 0x1b 0x4c 0xfd 0x0 0x0 0x4d 0xbd 0x0 0x0 0x0 0x1 0x0 0x44 0x5 0x3b 0x17 0x70 0x0 0x0 0xc6 0x72 0x0 0x0 0x0 0x0 0x0 0x2c 0x7a 0xaa 0x0 0x0 0x1a 0x62 0x0 0x0 0x0 0x0 0x6 0x1e 0x5 0x94 0x0 0x0 0x0 0x0 0x7 0xd0 0x0 0x0 0x59 0x32 0x0 0x1 0xf7 0x7e 0xff 0xff
2023-03-19 22:25:23.575 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x44 0x0 0x0 0x0 0xcb 0x1 0x3 0xc8 0x0 0x3 0x0 0x1 0x2 0x1 0x32 0x31 0x30 0x38 0x32 0x39 0x39 0x32 0x36 0x38 0x0 0x0 0xff 0xff 0x0 0x0 0x7 0x17 0x0 0x0 0x52 0x14 0x13 0x4e 0x0 0x0 0x38 0x80 0x0 0x1 0x2 0x3 0x0 0x0 0x0 0xff 0x0 0xa 0x17 0x3 0x13 0xf 0x18 0x30 0x0 0x1 0x1f 0x40 0xff 0xff 0x0 0x0 0x0 0x0 0x12 0xc0 0x0 0x0 0x27 0x10 0xb 0xb8 0x1f 0x40 0x7 0xd0 0x3 0x52 0x0 0x0 0x0 0x0 0x3 0xe8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x21 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0xff 0xda 0x0 0x0 0x0 0x0 0xe2 0xfb 0xff 0xff 0x13 0xc4 0x5 0xd6 0xfd 0xe6 0x43 0x8a 0x0 0x0 0x0 0x30 0x0 0x1b 0x4c 0xfd 0x0 0x0 0x4d 0xbd 0x0 0x0 0x0 0x1 0x0 0x44 0x5 0x3b 0x17 0x70 0x0 0x0 0xc6 0x72 0x0 0x0 0x0 0x0 0x0 0x2c 0x7a 0xaa 0x0 0x0 0x1a 0x62 0x0 0x0 0x0 0x0 0x6 0x1e 0x5 0x94 0x0 0x0 0x0 0x0 0x7 0xd0 0x0 0x0 0x59 0x32 0x0 0x1 0xf7 0x7e 0xff 0xff
2023-03-19 22:25:23.575 DEBUG (MainThread) [pymodbus.logging] Factory Response[0]
2023-03-19 22:25:23.575 ERROR (MainThread) [pymodbus.logging] Unable to decode response Modbus Error: Unknown response 0
2023-03-19 22:25:23.579 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/selector_events.py", line 876, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/home/vscode/.local/lib/python3.10/site-packages/pymodbus/client/base.py", line 268, in data_received
    self.framer.processIncomingPacket(data, self._handle_response, unit=0)
  File "/home/vscode/.local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 167, in processIncomingPacket
    self._process(callback)
  File "/home/vscode/.local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 186, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2023-03-19 22:25:23,583 INFO  logging:96 Protocol lost connection.

```

austinmroczek, 2023-03-19:
Same error with pymodbus 3.2.1

janiversen, 2023-03-20:
This is not a known problem, let me try to run in through our simulator and see if I can reproduce something.

Can  you please confirm that it always happens as the first request after a connect. It would be interesting to know why it disconnected, because that might be the real cause. Can you please supply a bit more log (last transaction that causes a disconnect together with the connect and the first transaction that it does not like).

I am thinking it could be some left-over from the preceding transaction that is causing a problem.

janiversen, 2023-03-20:
The message you send (00440000000601 / 0300000064) is (of course) a ReadHoldingRegistersRequest,
```
name            = ReadHoldingRegistersRequest
transaction_id  = 0x44
protocol_id     = 0x0
slave_id        = 0x1
skip_encode     = 0x0
check           = 0x0
address         = 0x0
count           = 0x64
```
It seems you are requesting 1024 registers not 100, this is a huge numbers and surely will cause problems from time to time.

Requesting 100 registers is a lot more normal.

janiversen, 2023-03-20:
The response is actually a valid response (0044000000cb01 / 03c8000300010201323130383239393236380000ffff0000071700005214134e0000388000010203000000ff000a1703130f183000011f40ffff0000000012c0000027100bb81f4007d003520000000003e800000000000000010000000100000000000000000000000000002134000000000000000000000002ffda00000000e2fbffff13c405d6fde6438a00000030001b4cfd00004dbd000000010044053b17700000c67200000000002c7aaa00001a6200000000061e05940000000007d0000059320001f77effff):
```
name            = ReadHoldingRegistersResponse
transaction_id  = 0x44
protocol_id     = 0x0
slave_id        = 0x1
skip_encode     = 0x0
check           = 0x0
registers       =
                . [3, 1, 513, 12849, 12344, 12857, 14642, 13880, 0, 65535, 0,
                . 1815, 0, 21012, 4942, 0, 14464, 1, 515, 0, 255, 10, 5891,
                . 4879, 6192, 1, 8000, 65535, 0, 0, 4800, 0, 10000, 3000,
                . 8000, 2000, 850, 0, 0, 1000, 0, 0, 0, 1, 0, 1, 0, 0, 0, 0,
                . 0, 0, 0, 8500, 0, 0, 0, 0, 0, 2, 65498, 0, 0, 58107, 65535,
                . 5060, 1494, 64998, 17290, 0, 48, 27, 19709, 0, 19901, 0, 1,
                . 68, 1339, 6000, 0, 50802, 0, 0, 44, 31402, 0, 6754, 0, 0,
                . 1566, 1428, 0, 0, 2000, 0, 22834, 1, 63358, 65535]
```

But you requested 1024 registers so therefore the frame is invalid.

However you need to look info why it reconnects, that is by far the more interesting problem.

austinmroczek, 2023-03-21:
I am building an HA integration for [NeoVolta](https://github.com/austinmroczek/neovolta), so I'll move toward calling the HA pymodbus integration.  Thanks for pointing that out and likely saving me lots of pain.

I'll post more logs in a bit (I closed out my environment and need to wait for errors again).

In my code the most registers I ever request at one time is 100, so I don't know where 1024 is coming from.  I can post the full code if you want, but basically I call:
 
```python
response = await self._get_value(0, 100)
# do stuff with response
response = await self._get_value(100, 100)
# do stuff with response

# nothing useful in the 200 range...

response = await self._get_value(300, 100)
# do stuff with response

```

I did some of testing of various different request sizes (requesting 100 registers 1 at a time, 10 at a time or 100 at a time), and 100 seemed to be the most efficient and least errors for this device.

The NeoVolta battery is connected to the network via an LSE-3 "stick logger."  NeoVolta (the company) provides access to data about my battery via the Solarman app.  I assume that means there is another modbus client embedded in the stick logger, which then sends data home to their server.  Is it possible two clients hammering the server is causing issues?

I will look into the connection/re-connect once I have errors in the log again.

austinmroczek, 2023-03-21:
Here are logs around a loss of connection:

```txt
...
2023-03-21 00:16:06.739 DEBUG (MainThread) [pymodbus.logging] Adding transaction 3
2023-03-21 00:16:07.195 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x3 0x0 0x0 0x0 0xcb 0x1 0x3 0xc8 0x8 0x1 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xc8 0x0 0xbc 0x0 0x10 0x0 0x0 0x0 0x10 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0x46 0x0 0x0 0x8e 0x1e 0x1f 0xb4 0x0 0x2c 0x8 0x0 0x14 0xec 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x5 0xac 0xc5 0x1 0xe1 0x0 0xf0 0xff 0xcf 0xff 0xde 0x0 0x1 0xff 0xfe 0x4 0xa1 0x4 0x9e 0x9 0x4f 0xff 0xa4 0x0 0xc8 0x13 0xcb 0x0 0x0 0x4 0x4b 0x7 0x17 0x0 0x2 0x9 0x3f 0x9 0x55 0x4 0xa1 0x4 0x9d 0x9 0x4e 0x9 0x57 0x4 0xa1 0x4 0x9d 0x9 0x3e 0x4 0xa1 0x4 0x9d 0x0 0x0 0x1 0x74 0x1 0x55 0x1 0x2a 0x0 0xd5 0x0 0x3c 0x0 0x5a 0x2 0x46 0xff 0xcf 0xff 0xde 0xff 0xad 0x0 0x0 0x0 0x0 0x0 0x0 0xff 0xd5 0x0 0x2b 0x0 0x0 0x0 0xc7 0x1 0x2c 0x1 0xf3 0x0 0xc7 0x1 0x2c 0x9 0x42 0x4 0x9b 0x14 0x94 0x0 0x63 0x1 0xf3 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x6d 0x0 0xd0 0x17 0x6d 0x17 0x6d 0x0 0x1 0x0 0x11 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-03-21 00:16:07.196 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x3 0x0 0x0 0x0 0xcb 0x1 0x3 0xc8 0x8 0x1 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xc8 0x0 0xbc 0x0 0x10 0x0 0x0 0x0 0x10 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0x46 0x0 0x0 0x8e 0x1e 0x1f 0xb4 0x0 0x2c 0x8 0x0 0x14 0xec 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x5 0xac 0xc5 0x1 0xe1 0x0 0xf0 0xff 0xcf 0xff 0xde 0x0 0x1 0xff 0xfe 0x4 0xa1 0x4 0x9e 0x9 0x4f 0xff 0xa4 0x0 0xc8 0x13 0xcb 0x0 0x0 0x4 0x4b 0x7 0x17 0x0 0x2 0x9 0x3f 0x9 0x55 0x4 0xa1 0x4 0x9d 0x9 0x4e 0x9 0x57 0x4 0xa1 0x4 0x9d 0x9 0x3e 0x4 0xa1 0x4 0x9d 0x0 0x0 0x1 0x74 0x1 0x55 0x1 0x2a 0x0 0xd5 0x0 0x3c 0x0 0x5a 0x2 0x46 0xff 0xcf 0xff 0xde 0xff 0xad 0x0 0x0 0x0 0x0 0x0 0x0 0xff 0xd5 0x0 0x2b 0x0 0x0 0x0 0xc7 0x1 0x2c 0x1 0xf3 0x0 0xc7 0x1 0x2c 0x9 0x42 0x4 0x9b 0x14 0x94 0x0 0x63 0x1 0xf3 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x6d 0x0 0xd0 0x17 0x6d 0x17 0x6d 0x0 0x1 0x0 0x11 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-03-21 00:16:07.196 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-03-21 00:16:07.196 DEBUG (MainThread) [pymodbus.logging] Getting transaction 3
2023-03-21 00:16:07.199 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x4 0x0 0x0 0x0 0x6 0x1 0x3 0x1 0x2c 0x0 0x64
2023-03-21 00:16:07.200 DEBUG (MainThread) [pymodbus.logging] Adding transaction 4
2023-03-21 00:16:10.214 INFO (MainThread) [pymodbus.logging] Protocol lost connection.
2023-03-21 00:16:10.216 DEBUG (MainThread) [pymodbus.logging] Client disconnected from modbus server: trying to send
2023-03-21 00:16:10.217 DEBUG (MainThread) [pymodbus.logging] Getting transaction 4
2023-03-21 00:16:10.217 DEBUG (MainThread) [custom_components.neovolta.api] Neovolta timeout: 
2023-03-21 00:16:10.217 DEBUG (MainThread) [custom_components.neovolta.api] Neovolta re-try # 2 for register 300
2023-03-21 00:16:10.219 INFO (MainThread) [pymodbus.logging] Protocol lost connection.
2023-03-21 00:16:10.220 WARNING (MainThread) [pymodbus.logging] Ignoring launch of delayed reconnection, another is in progress
2023-03-21 00:16:10.221 DEBUG (MainThread) [pymodbus.logging] Client disconnected from modbus server: None
2023-03-21 00:16:10.221 DEBUG (MainThread) [pymodbus.logging] Waiting 1000 ms before next connection attempt.
2023-03-21 00:16:11.226 DEBUG (MainThread) [pymodbus.logging] Connecting.
2023-03-21 00:16:11.241 DEBUG (MainThread) [pymodbus.logging] Client connected to modbus server
2023-03-21 00:16:11.241 INFO (MainThread) [pymodbus.logging] Protocol made connection.
2023-03-21 00:16:11.242 INFO (MainThread) [pymodbus.logging] Connected to 192.168.1.205:8899.
2023-03-21 00:16:12.920 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x3 0x0 0x0 0x0 0xcb 0x1 0x3 0xc8 0x8 0x1 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xc8 0x0 0xbc 0x0 0x10 0x0 0x0 0x0 0x10 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0x4b 0x0 0x0 0x8e 0x1e 0x1f 0xb3 0x0 0x2c 0x8 0x0 0x14 0xe6 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x5 0xac 0xc5 0x1 0xe1 0x0 0xf0 0xff 0xce 0xff 0xdd 0x0 0x1 0xff 0xff 0x4 0xa5 0x4 0xa2 0x9 0x56 0xff 0xa1 0x0 0xc8 0x13 0xca 0x0 0x0 0x4 0x4b 0x7 0x17 0x0 0x1 0x9 0x47 0x9 0x5d 0x4 0xa4 0x4 0xa0 0x9 0x54 0x9 0x5b 0x4 0xa4 0x4 0xa0 0x9 0x44 0x4 0xa4 0x4 0xa0 0x0 0x0 0x1 0x67 0x1 0x45 0x1 0x2c 0x0 0xd7 0x0 0x3c 0x0 0x5a 0x2 0x4c 0xff 0xcf 0xff 0xd2 0xff 0xa1 0x0 0x0 0x0 0x0 0x0 0x0 0xff 0xd3 0x0 0x21 0xff 0xf4 0x0 0xc8 0x1 0x19 0x1 0xe1 0x0 0xc8 0x1 0x19 0x9 0x4a 0x4 0x9b 0x14 0x92 0x0 0x63 0x1 0xe1 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x61 0x0 0xba 0x17 0x70 0x17 0x70 0x0 0x1 0x0 0x11 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-03-21 00:16:12.922 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x3 0x0 0x0 0x0 0xcb 0x1 0x3 0xc8 0x8 0x1 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xc8 0x0 0xbc 0x0 0x10 0x0 0x0 0x0 0x10 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0x4b 0x0 0x0 0x8e 0x1e 0x1f 0xb3 0x0 0x2c 0x8 0x0 0x14 0xe6 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x5 0xac 0xc5 0x1 0xe1 0x0 0xf0 0xff 0xce 0xff 0xdd 0x0 0x1 0xff 0xff 0x4 0xa5 0x4 0xa2 0x9 0x56 0xff 0xa1 0x0 0xc8 0x13 0xca 0x0 0x0 0x4 0x4b 0x7 0x17 0x0 0x1 0x9 0x47 0x9 0x5d 0x4 0xa4 0x4 0xa0 0x9 0x54 0x9 0x5b 0x4 0xa4 0x4 0xa0 0x9 0x44 0x4 0xa4 0x4 0xa0 0x0 0x0 0x1 0x67 0x1 0x45 0x1 0x2c 0x0 0xd7 0x0 0x3c 0x0 0x5a 0x2 0x4c 0xff 0xcf 0xff 0xd2 0xff 0xa1  [... cut ...]  0x0 0x0 0x0 0x61 0x0 0xba 0x17 0x70 0x17 0x70 0x0 0x1 0x0 0x11 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-03-21 00:16:12.923 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-03-21 00:16:12.923 DEBUG (MainThread) [pymodbus.logging] Getting transaction 3
2023-03-21 00:16:12.924 DEBUG (MainThread) [pymodbus.logging] Unrequested message: ReadHoldingRegistersResponse (100)
```

It looks like maybe I'm getting getting a response to transaction 4 but marked as 3.

austinmroczek, 2023-03-21:
Here's a much bigger chunk of logs that start showing the problem.

[home-assistant_2023-03-21T00-24-44.793Z.log](https://github.com/pymodbus-dev/pymodbus/files/11024014/home-assistant_2023-03-21T00-24-44.793Z.log)

austinmroczek, 2023-03-21:
Please let me know if you'd like to try to figure out the issue(s) now, or if you'd rather wait for me to move to using the HA pymodbus integration and then see if it's still a problem.

janiversen, 2023-03-21:
this is not a pymodbus issue, as far as I can see, you receive a timeout from the library, because we do not receive a response. This can easily be caused by the fact that you have 2 clients connected to the same device (most devices do not like that), and can possibly be avoided with a higher timeout.

The reconnect might cause your device to send the last response (again) and thus cause problems. We have never more than 1 outstanding transaction (because only very few devices allow multiple transactions).

A possible test is to configure the modbus integration to retrieve data and see if that causes problems.

janiversen, 2023-03-22:
Closing this as not a problem, feel free to add a comment if you disagree.

closed 2023-03-22 by janiversen

austinmroczek, 2023-03-24:
Thanks for the help.