pymodbus-dev/pymodbus #1454: pymodbus.server does not return any values

chkr-private, 2023-03-25:
### Versions

- Python: 3.10.10
- OS: Fedora 36
- Pymodbus: 3.2.2

### Pymodbus Specific

- Server: tcp
- Client: tcp

### Description

- starting pymodbus.server: ```pymodbus.server run```
- starting pymodbus.client: ```pymodbus.console tcp --host localhost --port 5020```
- when trying to read some registers, the following error appears after a short timeout
```
> client.read_input_registers address 1 count 1 slave 1
{
    "message": "Broadcast message, ignoring errors!!!"
}
````
- the same worked in pymodus 2.5.3

janiversen, 2023-03-25:
Hmmm that is something I have not tested.

@dhoomakethu could you please take a look at this. I might have broken something among the many changes.

janiversen, 2023-03-25:
Did you try with another address, I think the datastore for the server was updated.

dhoomakethu, 2023-03-26:
On the latest `master` branch I am hitting this issue. Will work on it.
```
❯ pymodbus.console tcp --host localhost --port 5020
Traceback (most recent call last):
  File "/Users/sanjay/.virtualenvs/pymodbus/bin/pymodbus.console", line 33, in <module>
    sys.exit(load_entry_point('pymodbus', 'console_scripts', 'pymodbus.console')())
  File "/Users/sanjay/.virtualenvs/pymodbus/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/Users/sanjay/.virtualenvs/pymodbus/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/Users/sanjay/.virtualenvs/pymodbus/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/sanjay/.virtualenvs/pymodbus/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/sanjay/.virtualenvs/pymodbus/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/Users/sanjay/.virtualenvs/pymodbus/lib/python3.8/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/sanjay/repo/pymodbus/pymodbus/repl/client/main.py", line 310, in tcp
    cli(client)
  File "/Users/sanjay/repo/pymodbus/pymodbus/repl/client/main.py", line 177, in cli
    completer=CmdCompleter(client),
  File "/Users/sanjay/repo/pymodbus/pymodbus/repl/client/completer.py", line 39, in __init__
    self._commands = commands or get_commands(client)
  File "/Users/sanjay/repo/pymodbus/pymodbus/repl/client/helper.py", line 214, in get_commands
    requests = _get_requests(members)
  File "/Users/sanjay/repo/pymodbus/pymodbus/repl/client/helper.py", line 166, in _get_requests
    commands = {
  File "/Users/sanjay/repo/pymodbus/pymodbus/repl/client/helper.py", line 167, in <dictcomp>
    f"client.{c[0]}": Command(
  File "/Users/sanjay/repo/pymodbus/pymodbus/repl/client/helper.py", line 79, in __init__
    self.param_help = self._create_arg_help()
  File "/Users/sanjay/repo/pymodbus/pymodbus/repl/client/helper.py", line 105, in _create_arg_help
    param, param_help = param.split(":param")[1].strip().split(":")
ValueError: too many values to unpack (expected 2)
```

dhoomakethu, 2023-03-26:
@chkr-private Please run your server with `pymodbus.server --verbose run` and paste the log here, I believe the the default unit id is `None` and which might be the reason why you are not receiving the values.
```
❯ pymodbus.server run --help

 Usage: pymodbus.server run [OPTIONS]

 Run Reactive Modbus server.
 Exposing REST endpoint for response manipulation.

╭─ Options ──────────────────────────────────────────────────────────────────────────────────────────────────────────────────╮
│ --modbus-server       -s      TEXT     Modbus Server [default: tcp]                                                        │
│ --framer              -f      TEXT     Modbus framer to use [default: socket]                                              │
│ --modbus-port         -p      INTEGER  Modbus port [default: 5020]                                                         │
│ --unit-id             -u      INTEGER  Supported Modbus unit id's [default: None]                                          │
│ --modbus-config-path          PATH     Path to additional modbus server config [default: None]                             │
│ --random              -r      INTEGER  Randomize every `r` reads. 0=never, 1=always,2=every-second-read, and so on.        │
│                                        Applicable IR and DI.                                                               │
│                                        [default: 0]                                                                        │
│ --change-rate         -c      INTEGER  Rate in % registers to change. 0=none, 100=all, 12=12% of registers, and so on.     │
│                                        Applicable IR and DI.                                                               │
│                                        [default: 0]                                                                        │
│ --help                                 Show this message and exit.                                                         │
╰────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────╯

```

dhoomakethu, 2023-03-26:
@chkr-private please give this PR a try. Thanks.

closed 2023-03-26 by janiversen

chkr-private, 2023-03-28:
> @chkr-private please give this PR a try. Thanks.

I just tested it at commit c77e8586c1946f477507e7ebffd94f868f91525f:

client:
```
$ ~/.local/bin/pymodbus.console tcp --host localhost --port 5020
----------------------------------------------------------------------------
__________          _____             .___  __________              .__
\______   \___.__. /     \   ____   __| _/  \______   \ ____ ______ |  |
 |     ___<   |  |/  \ /  \ /  _ \ / __ |    |       _// __ \\\____ \|  |
 |    |    \___  /    Y    (  <_> ) /_/ |    |    |   \  ___/|  |_> >  |__
 |____|    / ____\____|__  /\____/\____ | /\ |____|_  /\___  >   __/|____/
           \/            \/            \/ \/        \/     \/|__|
                                        v1.3.0 - 3.3.0.alpha
----------------------------------------------------------------------------

> client.read_input_registers address 3 count 1 slave 1
{
    "original_function_code": "4 (0x4)",
    "error": "[Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)"
}
```
server:
```
~/.local/bin/pymodbus.server --verbose run
2023-03-28 22:27:56,204 INFO  logging:96 Modbus server started
2023-03-28 22:27:56,207 INFO  logging:96 Server(TCP) listening.

__________                          .______.                    _________
\______   \___.__. _____   ____   __| _/\_ |__  __ __  ______  /   _____/ ______________  __ ___________
 |     ___<   |  |/     \ /  _ \ / __ |  | __ \|  |  \/  ___/  \_____  \_/ __ \_  __ \  \/ // __ \_  __ \\
 |    |    \___  |  Y Y  (  <_> ) /_/ |  | \_\ \  |  /\___ \   /        \  ___/|  | \/\   /\  ___/|  | \/
 |____|    / ____|__|_|  /\____/\____ |  |___  /____//____  > /_______  /\___  >__|    \_/  \___  >__|
           \/          \/            \/      \/           \/          \/     \/                 \/

SERVER > 2023-03-28 22:28:33,316 DEBUG logging:102 Socket [('::1', 5020)] opened
2023-03-28 22:28:33,316 DEBUG logging:102 TCP client connection established [('::1', 41878)]
2023-03-28 22:28:35,517 DEBUG logging:102 Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x3 0x0 0x1
2023-03-28 22:28:35,517 DEBUG logging:102 Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x3 0x0 0x1
2023-03-28 22:28:35,518 DEBUG logging:102 Not a valid slave id - 1, ignoring!!
2023-03-28 22:28:38,521 DEBUG logging:102 Handler for stream [('::1', 41878)] has been canceled
2023-03-28 22:28:38,521 DEBUG logging:102 TCP client disconnected [('::1', 41878)]
```

Ok, seems to indicate that the slave id supplied by the client is not recognized by the server.

I started now the pymodbus.server using the parameter: ```-u, --slave-id INTEGER     Supported Modbus slave id's```

```
~/.local/bin/pymodbus.server --verbose run -u 1
2023-03-28 22:32:10,583 INFO  logging:96 Modbus server started
2023-03-28 22:32:10,587 INFO  logging:96 Server(TCP) listening.

__________                          .______.                    _________
\______   \___.__. _____   ____   __| _/\_ |__  __ __  ______  /   _____/ ______________  __ ___________
 |     ___<   |  |/     \ /  _ \ / __ |  | __ \|  |  \/  ___/  \_____  \_/ __ \_  __ \  \/ // __ \_  __ \\
 |    |    \___  |  Y Y  (  <_> ) /_/ |  | \_\ \  |  /\___ \   /        \  ___/|  | \/\   /\  ___/|  | \/
 |____|    / ____|__|_|  /\____/\____ |  |___  /____//____  > /_______  /\___  >__|    \_/  \___  >__|
           \/          \/            \/      \/           \/          \/     \/                 \/

SERVER > 2023-03-28 22:32:14,822 DEBUG logging:102 Socket [('::1', 5020)] opened
2023-03-28 22:32:14,822 DEBUG logging:102 TCP client connection established [('::1', 50024)]
2023-03-28 22:32:17,535 DEBUG logging:102 Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x3 0x0 0x1
2023-03-28 22:32:17,535 DEBUG logging:102 Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x3 0x0 0x1
2023-03-28 22:32:17,535 DEBUG logging:102 Not a valid slave id - 1, ignoring!!
2023-03-28 22:32:20,538 DEBUG logging:102 Handler for stream [('::1', 50024)] has been canceled
2023-03-28 22:32:20,538 DEBUG logging:102 TCP client disconnected [('::1', 50024)]
```

That looks like that pymodbus.server does not recognize the parameter ```-u 1``` correctly either.

Most likely the reason is, that in pymodbus/repl/server/main.py, the type of "modbus_slave_id" is a tuple (which makes sense if someone would repeat the '-u' argument, but it also a tuple if there is just one) but in in pymodbus/server/reactive/main.py (create_context), the "isinstance(slave, list)" fails and so the tuple is wrapped as list again. Eventually, that will make the test in "_validate_slave_id" from pymodbus/framer/__init.py fail (```return self._header["uid"] in slaves```) (since slaves is now a tuple in a list like "[(1,)]").

It looks like, it pymodbus/framer/socket_framer.py, a similar code snipped is checking now for both, tuple and list:
```
        if not isinstance(slave, (list, tuple)):
            slave = [slave]
```

If I apply a similar patch to pymodbus/server/reactive/main.py, it does now accept the slave id via command line argument (and it also works without "-u 1" it, since "1" is the default).

```
git diff pymodbus/server/reactive/main.py 
diff --git a/pymodbus/server/reactive/main.py b/pymodbus/server/reactive/main.py
index dbbb3ab..6498ee8 100644
--- a/pymodbus/server/reactive/main.py
+++ b/pymodbus/server/reactive/main.py
@@ -411,7 +411,7 @@ class ReactiveServer:
         :return: ModbusServerContext object
         """
         data_block = data_block_settings.pop("data_block", DEFAULT_DATA_BLOCK)
-        if not isinstance(slave, list):
+        if not isinstance(slave, (list, tuple)):
             slave = [slave]
         slaves = {}
         for i in slave:

```

If you think it is the correct patch, I can create a PR.

dhoomakethu, 2023-03-29:
@chkr-private I double checked with latest `dev` and its working fine.

```
pymodbus on  dev [$?] via 🐍 v3.8.13 (pymodbus)
❯ pymodbus.server --verbose run -u 1
2023-03-29 10:59:42,227 INFO  logging:96 Modbus server started
2023-03-29 10:59:42,230 INFO  logging:96 Server(TCP) listening.

__________                          .______.                    _________
\______   \___.__. _____   ____   __| _/\_ |__  __ __  ______  /   _____/ ______________  __ ___________
 |     ___<   |  |/     \ /  _ \ / __ |  | __ \|  |  \/  ___/  \_____  \_/ __ \_  __ \  \/ // __ \_  __ \\
 |    |    \___  |  Y Y  (  <_> ) /_/ |  | \_\ \  |  /\___ \   /        \  ___/|  | \/\   /\  ___/|  | \/
 |____|    / ____|__|_|  /\____/\____ |  |___  /____//____  > /_______  /\___  >__|    \_/  \___  >__|
           \/          \/            \/      \/           \/          \/     \/                 \/

SERVER > 2023-03-29 10:59:49,914 DEBUG logging:102 Socket [('::1', 5020)] opened
2023-03-29 10:59:49,914 DEBUG logging:102 TCP client connection established [('::1', 62741)]
2023-03-29 10:59:53,140 DEBUG logging:102 Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x1 0x0 0x1
2023-03-29 10:59:53,140 DEBUG logging:102 Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x1 0x0 0x1
2023-03-29 10:59:53,140 DEBUG logging:102 Factory Request[ReadInputRegistersRequest': 4]
2023-03-29 10:59:53,140 DEBUG logging:102 validate: fc-[4] address-1: count-1
2023-03-29 10:59:53,140 DEBUG logging:102 getValues: fc-[4] address-1: count-1
```

chkr-private, 2023-03-29:
@dhoomakethu there must be some kind of difference in our setup. I just double-checked again that I can still reproduce the problem with origin/dev at commit f815003721f91496874d677b3a1255bbe00b6f53.

To verify my suspicion regarding the typer library, could you probably apply a patch like this and show the output of running ```pymodbus.server --verbose run -u 1```:

```
diff --git a/pymodbus/framer/__init__.py b/pymodbus/framer/__init__.py
index 949a60a..5f396c9 100644
--- a/pymodbus/framer/__init__.py
+++ b/pymodbus/framer/__init__.py
@@ -39,6 +39,7 @@ class ModbusFramer:
             # Handle Modbus TCP slave identifier (0x00 0r 0xFF)
             # in asynchronous requests
             return True
+        print("expected slaves", slaves)
         return self._header["uid"] in slaves  # pylint: disable=no-member
 
     def sendPacket(self, message):
```

In my case,  I get an output like this:
```
SERVER > 2023-03-30 00:20:54,659 DEBUG logging:102 Socket [('::1', 5020)] opened
2023-03-30 00:20:54,659 DEBUG logging:102 TCP client connection established [('::1', 48722)]
2023-03-30 00:20:55,658 DEBUG logging:102 Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x3 0x0 0x1
2023-03-30 00:20:55,658 DEBUG logging:102 Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x3 0x0 0x1
expected slaves [(1,)]
```
Which is consistent with my findings in comment https://github.com/pymodbus-dev/pymodbus/issues/1454#issuecomment-1487586835 .

If it is indeed related to the typer library, could you check which version is running in your system? For me, it is:
```
python3
Python 3.10.10 (main, Feb  8 2023, 00:00:00) [GCC 12.2.1 20221121 (Red Hat 12.2.1-4)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import typer
>>> typer.__version__
'0.4.1'
>>> 
```

dhoomakethu, 2023-03-30:
OK,. I see you are using python 3.10, let me see if it's Python version surviving specific.

chkr-private, 2023-03-30:
Thanks for the reply. On Fedora 36 the typer python library is still on version 0.4.1 which causes the problem. I can confirm that beginning with Fedora 37, typer is on 0.7.0 which does not have the problem anymore. So with that respect I think it would be OK not to add special handling for older typer versions in pymodbus...

That leaves just one question: would it make sense to change the default list of slave IDs for pymodbus.server from "none" to ID 1? That way the tool could be easier used without the need of specifying the ID.

PR #1456 (2023-03-26): Fix wrong error message with pymodbus console
Pass  `slave` info while processing exceptions when working with pymodbus repl client

```
pymodbus on  fix-#1454 [$?] via 🐍 v3.8.13 (pymodbus) took 2s
❯ pymodbus.console tcp --host localhost --port 5020

----------------------------------------------------------------------------
__________          _____             .___  __________              .__
\______   \___.__. /     \   ____   __| _/  \______   \ ____ ______ |  |
 |     ___<   |  |/  \ /  \ /  _ \ / __ |    |       _// __ \\\____ \|  |
 |    |    \___  /    Y    (  <_> ) /_/ |    |    |  [... cut]