pymodbus-dev/pymodbus #1460: specify Modbus NaN values

joanwa, 2023-03-27:
### Versions

- Python: 3.10
- OS: macOS
- Pymodbus: 3.3
- Modbus Hardware (if used): Any SMA Inverter

### Pymodbus Specific

- Server: tcp
- Client: tcp

### Description

The problem
I'm using modbus to read data from an PV inverter. Over night that inverter goes into sleep mode and only returns their declared NaN values, which becomes +-2,147,483,648 +-(2^32 /2) for an int32 value, depending on the register read.
The inverters manufacturer, SMA, provides a table with their NaN values (screenshot below).

It would be great if we can have an optional modbus parameter to specify its expected NaN return value. This saves a lot of h conditionals to show 'NaN' or 0 instead of the returned +-2,147,483,648.
[image]

janiversen, 2023-03-27:
This is not really something pymodbus cares about, it returns registers each 16bit and not values like float etc.

We do have convert functions, but I am not sure how such a change would fit. I think you would actually need a value filter to get rid of extreme small or extreme large values (The modbus integration in homeassistant uses filters).

Having said that, pull requests are welcome if you want a change.

janiversen, 2023-04-07:
@joanwa did you solve your problem ? (I can see you opened a PR in HomeAssistant, is it because this is not a pymodbus problem ?)

janiversen, 2023-04-07:
Closing this issue as not a problem in pymodbus.

closed 2023-04-07 by janiversen

joanwa, 2023-04-07:
Yes exactly, can be solved in home assistant, not a pymodbus issue! Thanks!