pymodbus-dev/pymodbus #1464: ModbusIOException pymodbus exception returned instead of being raised.

giacomonicolini, 2023-03-30:
### Versions

* Python: 3.11
* OS: Debian GNU/Linux
* Pymodbus: 3.2.2

### Pymodbus Specific
* Client: tcp - sync

### Description

I am using the pymodbus library to interconnect with an inverter and read data using ModbusTCP. The data is read every 10 seconds and sometimes it happens that connection errors occur. In these cases I encounter two types of exceptions: 
- pymodbus.exceptions.ConnectionException
- pymodbus.exceptions.ModbusIOException

In the case of the ModbusIOException, this is returned by the address read function instead of being raised. This forces me to do an additional check before I can read the values into the registers, specifically checking that the returned value is not of type ModbusIOException instead of handling everything with a simple try, except (with ModbusIOException as exception). In the source code in fact, I could see that exceptions are not raised, but returned and attributed to the value. I would suggest raising the exception instead of returning them.

### Code and Logs
-- HOW DO I USE IT
```python
from pymodbus.client import ModbusTcpClient
from pymodbus.exceptions import ConnectionException, ModbusIOException

def get_data(address, port):
  client = ModbusTcpClient(address, port)
  try:
      data = self._client.read_holding_registers(
          address=starting_addr, count=count, slave=1
      )
  except (ConnectionException, ModbusIOException) as e:
      print(e)
  return data.registers
```

-- WHICH ERROR DOES IT THROW ME:
```
AttributeError: 'ModbusIOException' object has no attribute 'registers'
```
-- HOW DO I HANDLE IT NOW:
```python
from pymodbus.client import ModbusTcpClient
from pymodbus.exceptions import ConnectionException, ModbusIOException

def get_data(address, port):
  client = ModbusTcpClient(address, port)
  try:
      data = self._client.read_holding_registers(
          address=starting_addr, count=count, slave=1
      )
  except ConnectionException as e:
      print(e)
  if isinstance(data, ModbusIOException):
      print(data)
      return
  return data.registers
```
-- THE pymodbus SOURCE CODE:
```python
class ReadHoldingRegistersRequest(ReadRegistersRequestBase):
    """Read holding registers.

    This function code is used to read the contents of a contiguous block
    of holding registers in a remote device. The Request PDU specifies the
    starting register address and the number of registers. In the PDU
    Registers are addressed starting at zero. Therefore registers numbered
    1-16 are addressed as 0-15.
    """

    function_code = 3

    def __init__(self, address=None, count=None, unit=Defaults.Slave, **kwargs):
        """Initialize a new instance of the request.

        :param address: The starting address to read from
        :param count: The number of registers to read from address
        :param unit: Modbus slave unit ID
        """
        super().__init__(address, count, unit, **kwargs)

    def execute(self, context):
        """Run a read holding request against a datastore.

        :param context: The datastore to request from
        :returns: An initialized :py:class:`~pymodbus.register_read_message.ReadHoldingRegistersResponse`, or an :py:class:`~pymodbus.pdu.ExceptionResponse` if an error occurred
        """
        if not (1 <= self.count <= 0x7D):
            return self.doException(merror.IllegalValue)
        if not context.validate(self.function_code, self.address, self.count):
            return self.doException(merror.IllegalAddress)
        values = context.getValues(self.function_code, self.address, self.count)
        return ReadHoldingRegistersResponse(values)
```
```python
class ModbusRequest(ModbusPDU):
    """Base class for a modbus request PDU."""

    function_code = -1

    def __init__(self, unit=Defaults.Slave, **kwargs):
        """Proxy to the lower level initializer.

        :param unit: Modbus slave unit ID
        """
        super().__init__(unit, **kwargs)

    def doException(self, exception):  # pylint: disable=invalid-name
        """Build an error response based on the function.

        :param exception: The exception to return
        :raises: An exception response
        """
        exc = ExceptionResponse(self.function_code, exception)
        _logger.error(exc)
        return exc
```

janiversen, 2023-03-30:
Exceptions raised originates in the library while exceptions returned are actual modbus messages, so not a python exception, but a device exception.

If you want it to work differently feel free to submit a PR, do not forget to update documentation and tests.

janiversen, 2023-04-07:
Closing a not an error.

closed 2023-04-07 by janiversen