pymodbus-dev/pymodbus #1466: AsyncModbusTcpClient: Cannot read after one failed read

02JanDal, 2023-04-01:
### Versions

- Python: 3.10.9
- OS: Ubuntu 20.04 in WSL2 in Windows 11
- Pymodbus: 3.2.2
- Modbus Hardware (if used): Sungrow SH10RT (solar inverter) with and without WiNet-S dongle

### Pymodbus Specific

- ~~Server: tcp/rtu/ascii - sync/async~~
- Client: **tcp**/rtu/ascii - sync/**async**

### Description

Depending on firmware versions and connection type involved, the set of supported registers varies. Therefore, I'm trying to write some code to detect which registers can be used. The most obvious, calling `read_input_registers` on each register and catching `ModbusException`s fails by subsequent registers after the failing also failing, even if they should succeed because the failure causes pymodbus to disconnect. Furthermore, calling `await client.close()` seems to keep some internal state which makes further calls not work, only recreating the client object from scratch works.

As such, I have a working workaround, however it feels a little hacky. Though I don't understand the Modbus protocol sufficiently to say so with confidence, but I suspect that the root cause might be the server in question misbehaving, however, it would be nice if pymodbus could handle this gracefully (either by automatically reconnecting or at least working after a manual re-connect).

### Code and Logs

First attempt:

```python
import asyncio
import sys

from pymodbus.client import AsyncModbusTcpClient
from pymodbus.exceptions import ModbusException
import pymodbus

pymodbus.pymodbus_apply_logging_config()

async def main():
    client = AsyncModbusTcpClient("192.168.3.20", 502)
    await client.connect()
    for address, count in [(4949, 2), (4951, 2), (4953, 15), (4968, 15), (4999, 1)]:
        try:
            print("Getting", address, file=sys.stderr)
            res = await client.read_input_registers(address, count, slave=1)
            print(res.registers, file=sys.stderr)
        except ModbusException as e:
            print(e, file=sys.stderr)

if __name__ == "__main__":
    asyncio.run(main())
```

<details><summary>Log</summary>
```
2023-03-31 05:12:48,976 DEBUG logging:102 Connecting to 192.168.3.20:502.
2023-03-31 05:12:48,976 DEBUG logging:102 Connecting.
2023-03-31 05:12:48,989 DEBUG logging:102 Client connected to modbus server
2023-03-31 05:12:48,989 INFO  logging:96 Protocol made connection.
2023-03-31 05:12:48,989 INFO  logging:96 Connected to 192.168.3.20:502.
Getting 4949
2023-03-31 05:12:48,989 DEBUG logging:102 send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x55 0x0 0x2
2023-03-31 05:12:48,989 DEBUG logging:102 Adding transaction 1
2023-03-31 05:12:48,999 DEBUG logging:102 recv: 0x0 0x1 0x0 0x0 0x0 0x7 0x1 0x4 0x4 0x31 0x30 0x41 0x41
2023-03-31 05:12:49,000 DEBUG logging:102 Processing: 0x0 0x1 0x0 0x0 0x0 0x7 0x1 0x4 0x4 0x31 0x30 0x41 0x41
2023-03-31 05:12:49,000 DEBUG logging:102 Factory Response[ReadInputRegistersResponse': 4]
2023-03-31 05:12:49,000 DEBUG logging:102 Getting transaction 1
[12592, 16705]
Getting 4951
2023-03-31 05:12:49,000 DEBUG logging:102 send: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x57 0x0 0x2
2023-03-31 05:12:49,000 DEBUG logging:102 Adding transaction 2
2023-03-31 05:12:49,007 DEBUG logging:102 recv: 0x0 0x2 0x0 0x0 0x0 0x7 0x1 0x4 0x4 0x4 0x0 0x1 0x0
2023-03-31 05:12:49,007 DEBUG logging:102 Processing: 0x0 0x2 0x0 0x0 0x0 0x7 0x1 0x4 0x4 0x4 0x0 0x1 0x0
2023-03-31 05:12:49,007 DEBUG logging:102 Factory Response[ReadInputRegistersResponse': 4]
2023-03-31 05:12:49,007 DEBUG logging:102 Getting transaction 2
[1024, 256]
[... 274 lines cut ...]
  File "/home/jan/projects/homeassistant-core/venv/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 167, in processIncomingPacket
    self._process(callback)
  File "/home/jan/projects/homeassistant-core/venv/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 186, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2023-03-31 05:18:10,054 INFO  logging:96 Protocol lost connection.
2023-03-31 05:18:10,054 DEBUG logging:102 Client disconnected from modbus server: Modbus Error: [Input/Output] Unable to decode request
2023-03-31 05:18:10,055 DEBUG logging:102 Getting transaction 1
2023-03-31 05:18:10,055 DEBUG logging:102 Waiting 100 ms before next connection attempt.
Modbus Error: [Connection] Connection lost during request
2023-03-31 05:18:10,055 DEBUG logging:102 Connecting to 192.168.3.20:502.
2023-03-31 05:18:10,055 DEBUG logging:102 Connecting.
2023-03-31 05:18:10,063 DEBUG logging:102 Client connected to modbus server
2023-03-31 05:18:10,063 INFO  logging:96 Protocol made connection.
2023-03-31 05:18:10,063 INFO  logging:96 Connected to 192.168.3.20:502.
Getting 4999
2023-03-31 05:18:10,063 DEBUG logging:102 send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x13 0x87 0x0 0x1
2023-03-31 05:18:10,063 DEBUG logging:102 Adding transaction 1
2023-03-31 05:18:10,075 DEBUG logging:102 recv: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0xe 0x3
2023-03-31 05:18:10,076 DEBUG logging:102 Processing: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0xe 0x3
2023-03-31 05:18:10,076 DEBUG logging:102 Factory Response[ReadInputRegistersResponse': 4]
2023-03-31 05:18:10,076 DEBUG logging:102 Getting transaction 1
[3587]
```
</details>

janiversen, 2023-04-02:
looks like you found a problem, I will write a test case to verify it.

janiversen, 2023-04-02:
The client do reconnect automatically but the app is notified because it might need some handling of the failed call.

janiversen, 2023-04-07:
Checked an extra time, reconnect automatically do work, however if you do not use the (relatively) new callback, you do not know the state of the connection.

We are in general in the process of refactoring the whole client, because there are a number of "strange" cases, like the reconnect.

PR #1475 (2023-04-07): Test reuse of client object

fixes #1466

The async client object can be reused as shown in the new test case.
closed 2023-04-07 by janiversen