pymodbus-dev/pymodbus #1468: ModbusUdpClient can get into a state where it returns responses from earlier requests

wirehell, 2023-04-03:
### Versions

- Python: 3.10.6
- OS: Ubuntu 22.04 LTS
- Pymodbus: 3.2.2
- Modbus Hardware (if used): -

### Pymodbus Specific

Client: udp - sync

### Description

After a duplicate UDP packet (and possible late data), stale data in network buffers will be attached to next request resulting in corrupted reads. This error state will persist and you will always get data from previous request until socket is reset.

I believe this is due to:
https://github.com/pymodbus-dev/pymodbus/blob/dev/pymodbus/transaction.py#L170
https://github.com/pymodbus-dev/pymodbus/blob/dev/pymodbus/transaction.py#L215-L222
Where when a stale response is received, it is matched against an earlier request, instead of the current. 
With the side effect of leaving the current response for the next request.

I've verified with Wireshark that the that that things are sent correctly over network.

I would expect for a sync client, that the response matches the current request unless there is an error (which shouldn't affect future requests).

It could be possible that this could affect other clients in some cases, but I have not looked into that.

### Code and Logs

```python
>>> from pymodbus.client import ModbusUdpClient
>>> s = ModbusUdpClient("2001:db8:100::1")
>>> s.connect()
True
>>> s.read_input_registers(2080,2,1).registers 
[5, 27314]
>>> for i in range(100): s.read_input_registers(2048,7,1).registers
```
CTRL+C here yields a high probability that there is an outstanding UDP request in the receive buffer for the client.
```
s.read_input_registers(2080,2,1).registers 
[14128, 12337, 12337, 12337, 12597, 13617, 22272]
>>> s.read_input_registers(2080,2,1).registers 
[5, 47802]
```

wirehell, 2023-04-03:
I wrote a test-case for this. Not a good test, but it shows the issue in a reproducible way:
https://github.com/pymodbus-dev/pymodbus/compare/dev...wirehell:pymodbus:udp-issue?expand=1

Output:
```
===================================================================================== FAILURES =====================================================================================
_______________________________________________________________ TestSynchronousClient.test_udp_client_recv_duplicate _______________________________________________________________
[gw0] linux -- Python 3.10.6 /home/kenny/src/pymodbus-dev/bin/python3

self = <test.test_client_sync.TestSynchronousClient object at 0x7ffff51a9540>

    def test_udp_client_recv_duplicate(self):
        """Test the udp client receive method"""
        client = ModbusUdpClient("127.0.0.1")
    
        client.socket = mockSocket2()
        client.socket.mock_prepare_receive(b"\x00\x01\x00\x00\x00\x05\x01\x04\x02\x00\x03");            # Response 1
        reply1 = client.read_input_registers(0x820, 1, 1)
        client.socket.mock_prepare_receive(b"\x00\x01\x00\x00\x00\x05\x01\x04\x02\x00\x03");            # Duplicate response 1
        client.socket.mock_prepare_receive(b"\x00\x02\x00\x00\x00\x07\x01\x04\x04\x00\x03\xf6\x3e")     # Response 2
        reply2 = client.read_input_registers(0x820, 2, 1)
        reply3 = client.read_input_registers(0x820, 100, 1)
    
        print(reply1.registers)
        print(reply2.registers)
        print(reply3.registers)
        print(reply1.transaction_id)
        print(reply2.transaction_id)
        print(reply3.transaction_id)
    
>       assert 1 == 0
E       assert 1 == 0

test/test_client_sync.py:102: AssertionError
------------------------------------------------------------------------------- Captured stdout call -------------------------------------------------------------------------------
[3]
[3]
[3, 63038]
1
1
2
============================================================================= short test summary info ==============================================================================
FAILED test/test_client_sync.py::TestSynchronousClient::test_udp_client_recv_duplicate - assert 1 == 0

```

janiversen, 2023-04-04:
Looks like you stumbled over a problem the code did not foresee. I have not looked at the code that handles transactions but if my memory serves me well, we do not check for duplicates.

wirehell, 2023-04-04:
At a glance, it looks like the async client does the right thing. 
Have you considered implementing the sync clients as blocking calls to the async clients? 
I think that would simplify things, i.e. less code to maintain, most of the transaction.py could be removed. 
Drawbacks would be that retries seems to be missing in async case, so that would need to be added. 
There may be other things that I'm not aware of as well since I've only looked at a part of the code and don't have much project context.

janiversen, 2023-04-04:
The PR clean client class is just about that, but the main problem is mixin.py which only have one set of calls used from both async and sync.

janiversen, 2023-04-08:
Thanks to your test case (slightly modified) I can now reproduce the problem.

However I am unsure about one thing, you state that the async client works, this makes me wonder because the transaction code is shared.

PR #1486 (2023-04-13): Duplicate transactions in UDP.
fixes #1468
closed 2023-04-13 by janiversen