pymodbus-dev/pymodbus #1472: Simulator returns IllegalAddress on read_coil functions using the correct register as address

corollaries, 2023-04-06:
### Versions

- Python: 3.10.6
- OS: Pop!_OS 22.04 LTS
- Pymodbus: 3.1.3

### Pymodbus Specific

- Server: Any
- Client: Any

### Description

I have configured registers with address 3 and 4 under `"bits"` and trying to execute a `READ_COIL` request on these specific registers, but keep on getting `IllegalAddress` as a response. After inspecting the simulator code I encountered the following line: [https://github.com/pymodbus-dev/pymodbus/blob/f815003721f91496874d677b3a1255bbe00b6f53/pymodbus/datastore/simulator.py#L556](https://github.com/pymodbus-dev/pymodbus/blob/f815003721f91496874d677b3a1255bbe00b6f53/pymodbus/datastore/simulator.py#L556) What is the purpose of dividing the address by 16? Essentially this line means that my address 3 gets turned into 0 which is why I'm getting the `IllegalAddress` exception. 

This essentially means that when invoking `READ_COIL` with address 48, everything does work. Based on the PR description that introduced this change: [https://github.com/pymodbus-dev/pymodbus/pull/1195](https://github.com/pymodbus-dev/pymodbus/pull/1195) it should be possible to read an address using both `read_coil` and `read_holding_register` but using the same address for both of these functions will result in `read_coil` failing. 

The sample code I've provided on the bottom should show this situation. Would love to know if this behavior is expected and/or there is possibly a bug needed to be fixed here. 

### Code and Logs

Client code:
```python
import time

from pymodbus.client import ModbusTcpClient
client = ModbusTcpClient('127.0.0.1', port=5022)

connected = client.connect()

if not connected:
    print("Connection failed")
    exit(1)

while True:
    response = client.read_coils(3, 1)
    if response.isError():
        print(f"Error occurred: {response}")
    else:
        print(f"Response: {response}")

    response = client.read_holding_registers(1, 1)
    if response.isError():
        print(f"Error occurred: {response}")
    else:
        print(f"Response: {response}")
    time.sleep(2)
```

Simulator code:
```python
import asyncio
from pymodbus.server import StartAsyncTcpServer
from pymodbus.datastore import ModbusServerContext, ModbusSimulatorContext

config = {
    "setup": {
        "co size": 250,
        "di size": 250,
        "hr size": 250,
        "ir size": 250,
        "shared blocks": True,
        "type exception": False,
        "defaults": {
            "value": {
                "bits": 0,
                "uint16": 0,
                "uint32": 0,
                "float32": 0,
                "string": "",
            },
            "action": {
                "bits": None,
                "uint16": None,
                "uint32": None,
                "float32": None,
                "string": None,
            },
        },
    },
    "invalid": [
    ],
    "bits": [
        {
            "addr": 3,
            "value": 0
        },
        {
            "addr": 4,
            "value": 0,
        }
    ],
    "uint16": [
        {
            "addr": 1,
            "value": 0
        },
        {
            "addr": 2,
            "value": 0,
        }
    ],
    "write": [
        2, 4
    ],
    "uint32": [
    ],
    "float32": [
    ],
    "string": [
    ],
    "repeat": [
    ],
}

slave = ModbusSimulatorContext(config, {})
server_context = ModbusServerContext(slaves=slave, single=True)
asyncio.run(StartAsyncTcpServer(server_context, None, ('127.0.0.1', 5022), defer_start=False))
```

Client logs:
```
Error occurred: Exception Response(129, 1, IllegalAddress)
Response: ReadHoldingRegistersResponse (1)
Error occurred: Exception Response(129, 1, IllegalAddress)
Response: ReadHoldingRegistersResponse (1)
Error occurred: Exception Response(129, 1, IllegalAddress)
Response: ReadHoldingRegistersResponse (1)
Error occurred: Exception Response(129, 1, IllegalAddress)
Response: ReadHoldingRegistersResponse (1)
```

Simulator logs:
```
ERROR:pymodbus.logging:Exception response Exception Response(129, 1, IllegalAddress)
ERROR:pymodbus.logging:Exception response Exception Response(129, 1, IllegalAddress)
ERROR:pymodbus.logging:Exception response Exception Response(129, 1, IllegalAddress)
```

janiversen, 2023-04-06:
It really depend on how you define the register storage, if it is shared read_coils reads register 0 for addresses 0 to 15, and register 1 for addresses 16 to 31.

If you are not using shared registers, each coil register represent one bit so address == bit.

It might be confusing but it is not our invention, but merely the modbus standard.

corollaries, 2023-04-07:
Thanks, that explains a few things. Maybe a stupid question but if we're not using shared blocks, shouldn't it be possible to set the following configuration for uint16 and bits?
```json 
    "bits": [
        {
            "addr": 0,
            "value": 0
        },
        {
            "addr": 1,
            "value": 0,
        }
    ],
    "uint16": [
        {
            "addr": 0,
            "value": 0
        },
        {
            "addr": 1,
            "value": 0,
        }
    ],
```
In the end it means that `uint16` addresses versus `bits` mean something different and should point to a different location inside a different block, right?  The simulator throws a RuntimeError on this however:
```
    raise RuntimeError(f'ERROR "{Label.type_uint16}" {reg} used')
RuntimeError: ERROR "uint16" <pymodbus.datastore.simulator.Cell object at 0x7f5c64163f10> used
```

I tried various things and this case particularly made me wrongfully conclude the addresses are one and the same for both. In any case, we can work with this after your explanation, thanks!

janiversen, 2023-04-07:
That should work, let me take a look.

janiversen, 2023-04-07:
Confirmed its a bug.

janiversen, 2023-04-07:
Found the problem. The registers represent physical addresses in memory, this is easy with shared=true, because all addresses are identical.

With shared=false, int16 is in the second block and thus you need to add the size of the CO block, making the addresses being 10,11 if size_ci == 10. Please remember that coils can only be in the first block, but the rest can be in any of the 3 other blocks.

We tried to have 4 section names for each type (int16_co, int16_di ...) but that became a lot more complicated to configure.

closed 2023-04-07 by janiversen

janiversen, 2023-04-11:
That is because 0 is not a valid register address, we do however still define it in the storage.

corollaries, 2023-04-11:
Thanks for the reply! How are we supposed to read address 0 to 15 in this case? If we try to perform `read_coil` on for example address 1. The same `validate` function divides this by `16` to get the `real_address`, which in this example will be `0`. 
https://github.com/pymodbus-dev/pymodbus/blob/6f7b66ed204f7c9d05833e7e65ab7819d8c7d476/pymodbus/datastore/simulator.py#L556 only to throw an `InvalidAddress` a couple lines later. 

Essentially this means that for `shared blocks = True` the read_coils will only start working from address 16 onwards. 
Correct me if I'm wrong but it seems that the standard says registers are addressed starting at zero, which means it should be a valid address?

janiversen, 2023-04-11:
In a shared mode you cannot and in the real world no device would define register 0 as coils or other type, so your question is rather theoretically. Some very old devices use register 0 as holding register, and the client have an option to facilitate that.

Registers begin with 1 but of course bit addresses begin with 0 (from register 1 and onwards).

janiversen, 2023-04-11:
or even better submit a PR which we can review.

PR #1474 (2023-04-07): Comment about addressing when shared=false

fixes #1472