pymodbus-dev/pymodbus #1477: Evaluate using HomeAssistant and bandit rules in ruff

alexrudd2, 2023-04-07:
If it makes things simpler then please make a PR that changes from bandit to flake8-bandit.

Can you please ( if you have not already) the rules from HA, commented out, that way it is easier to activate when you find t8mr.

_Originally posted by @janiversen in https://github.com/pymodbus-dev/pymodbus/issues/1476#issuecomment-1500557392_

janiversen, 2023-04-07:
I did not express myself well, please add these rules in our ruff.toml but comment those out that currently gives you a headache.

This PR is clean and nice, let's keep it like that, and then you can un comment the rules in new PR(s).

alexrudd2, 2023-04-07:
> I did not express myself well, please add these rules in our ruff.toml but comment those out that currently gives you a headache.

I understood you, but remember better if it's an issue assigned to me.  (I work across a dozen projects)

janiversen, 2023-04-07:
Ahh it is a new one, sorry I am on my iPAD this evening.

janiversen, 2023-04-08:
Looking forward to see more PR around this theme. 

I too work on a number of projects and I have absolutely no problem with you opening issues as reminders.

alexrudd2, 2023-04-10:
See #1480 and #1481

janiversen, 2023-04-10:
Looks like you have the version active ? or am I missing something?

alexrudd2, 2023-04-11:
> Looks like you have the version active ? or am I missing something?

I don't understand your question.

===

Anyways, I've gone and deleted some rules where I don't see value (such as `flake8-print`), and enabled a few more easy ones.

There are three remaining "big" rulesets which I think are useful, but require more substantial work.  I've put them in but commented out.
- bandit
- Tryceratops
- McCabe complexity

janiversen, 2023-04-11:
we already use bandit directly and mccabe via pylint, so it puzzles me why those 2 are big jobs....the third one is new to me.

Anyhow if you want we can divide them, or you do it slowly when you have time.

alexrudd2, 2023-04-11:
> Bandit is used:
> 

o_0. OK, I missed that.  Hopefully https://github.com/pymodbus-dev/pymodbus/pull/1488 makes it clearer (and kills another CI runner, yay)

However, `ruff` still doesn't actually have full parity with `flake8-bandit` so I think it's best to wait a bit.  (https://github.com/charliermarsh/ruff/issues/1646)

> Reducing [the McCabe complexity] is a nice goal, but please please do it very carefully, and in chunks I can review....I have already tried and failed (or at least postponed).

Agreed, it's difficult!
[image]

janiversen, 2023-04-11:
I believe you do !

janiversen, 2023-04-14:
I just read about ruff, in another project. Is rule M001 default ? if not we should activate it, to get warnings when there are unused noga statements.

alexrudd2, 2023-04-17:
Hmm, it appears `M001` has been remapped to `RUF100`. This would be part of the `RUF` rules, which are not enabled by default, BUT are enabled for pymodbus [here](https://github.com/pymodbus-dev/pymodbus/blob/dev/ruff.toml#L28).

I'd suggest moving the other project from `M001` to `RUF100` if they haven't already.

alexrudd2, 2023-04-24:
> > we already use bandit directly and mccabe via pylint, so it puzzles me why those 2 are big jobs....the third one is new to me.
> 
> For McCabe/pylint, I intend to actually reduce the complexity and not just use `pylint: ignore` :)

If https://github.com/pymodbus-dev/pymodbus/pull/1510 is OK, we're down to two functions with a McCabe complexity of > 10.

> pymodbus/transaction.py:127:9: C901 `execute` is too complex (23 > 10)
> pymodbus/transaction.py:335:9: C901 `_recv` is too complex (19 > 10)

Note that both of them have large `if/else if/else` blocks that are functionally switch statements.  If 
https://github.com/charliermarsh/ruff/issues/3597 gets implemented that will reduce their scores.

janiversen, 2023-04-24:
I have reworked many if/else constructs to
if - return...and no else. I try in general not to have nested if/else.

So I suggest you follow the same pattern if possible.

alexrudd2, 2023-04-24:
> I have reworked many if/else constructs to if - return...and no else. I try in general not to have nested if/else.
> 
> So I suggest you follow the same pattern if possible.

Yes, it's good advice when possible for _nested_ if/else.  However, I meant code like this (sequential if / else if / else) that would be a switch/case or match in other languages:
https://github.com/pymodbus-dev/pymodbus/blob/759b222869f5da4ce26d2d18be826bf258ed7a0e/pymodbus/transaction.py#L340-L349

Actually, since there are now only a few left I think I will just ignore them, switch to ruff, and disable the pylint McCabe.

janiversen, 2023-04-24:
Switch to ruff sounds like a good idea, but please do not disable McCabe, but add "pylint: disable=complex..." or whatever it is called in ruff, for those couple of methods.

alexrudd2, 2023-04-24:
> Switch to ruff sounds like a good idea, but please do not disable McCabe, but add "pylint: disable=complex..." or whatever it is called in ruff, for those couple of methods.

I meant disabling the _pylint_ McCade while enabling it in ruff.  https://github.com/pymodbus-dev/pymodbus/pull/1510/commits/a25de7fad942b921a0715949d09488795b4a7692

PR #1514 (2023-04-25): Enable remaining HomeAssistant rules in ruff
Closes #1477

There are still more rules to enable, and `ignore`s to remove, but I'm happy with this for now.
closed 2023-04-25 by janiversen