pymodbus-dev/pymodbus #1490: 3.2.2_master vs 3.2.2_dev

hhoefling, 2023-04-12:
Pip3 installs the 3.2.2(dev) on python3.9 on debian bullseye
This breaks my code.
Why  is Param "unit" renamed to "slave" ?

janiversen, 2023-04-12:
This does not seem to be a bug in pymodbus, so closing it.

It is normal that new versions have changes, why else make a new version.

closed 2023-04-12 by janiversen

hhoefling, 2023-04-12:
These Changes is notet for 3.3.0 not for 3.2.2!

janiversen, 2023-04-12:
Yes in 3.3 the change is completed, but it started in 3.0. For 3.0, 3.1 and 3.2 there are changes in some calls as noted. 3.2.2 do not have API changes since it is a point release that only solves bugs.

If you want changes feel free to submit a pull request.

janiversen, 2023-04-12:
See v3.1.0:

unit= in client calls are no longer converted to slave=, but raises a runtime exception.

That is most likely your problem.

hhoefling, 2023-04-12:
Ok, then i must use onother lib.
I will have same source for python3.5 and python3.9
(python3.5 maximal pymodbus 2.5.3)
(python3.9 minimal pymodbus 3.x is not compatieble)

janiversen, 2023-04-12:
I have been living with this sort of version differences most of my life, so I am used to have checks on python.version in the code....which sounds like less work, than to adapt to a new library. But of course that is completely your choice.

hhoefling, 2023-04-13:
> This doesn't seem to be a bug in Pymodbus, so close it.

I think this is a bug in 3.2.2
The change for "unit->slave" is backported from 3..0dev to 3.2.2
But now I have no way to get a 3.2.2 with old "unit" support.

My scripts are mostly 10 liners withch read only one modbus registers. The Scripts runs with pymodbus since 1.5.x
But now..... Bad..
I don't see any reason for the change.
No code reuse.
No promotion for open source,

hhoefling, 2023-04-13:
@janiversen 
you wrote on #1427.

> 
> From version 3.3 (see dev.) unit= is gone.
> 
After that these changes are backportet to 3.2.2

janiversen, 2023-04-13:
no they are not, that is the changes from 3.1

janiversen, 2023-04-13:
Changes in 3.2.2

https://github.com/pymodbus-dev/pymodbus/blob/dev/CHANGELOG.rst#version-322-picked-from-dev-only-bugfixes

Changes in 3.2.1

https://github.com/pymodbus-dev/pymodbus/blob/dev/CHANGELOG.rst#version-321-picked-from-dev-only-bugfixes

and just for the record we never backport, our commits move forward in a straight line. The reason being it is a lot easier to maintain an app without backports.

janiversen, 2023-04-13:
Just 2 thoughts to finalize this theme:

between v1.5 and v2.5.3 there have been significant API changes, so you are already used to change your app.

If v2.5.3 works for you, then why not simply continue using it, no need to upgrade. I suspect you in reality is upgrading from 2.5.3 and there you are very likely too see other differences, e.g. the return classes have changed between 2.5.3 and 3.0.0.

janiversen, 2023-04-13:
I assume you do know that you can just do:

pip3 install pymodbus==2.5.3 

and you get that version.

hhoefling, 2023-04-13:
Thanks for the hint.
Through my troubleshooting in the last few hours, I've found that too.
So I just have to adjust the setup.
because I can no longer use the default version of the operating system.