pymodbus-dev/pymodbus #1517: Update repo security.

janiversen, 2023-04-25:
Something is configured wrong in security, committers (people with write bit) should be able to merge a PR.

Demand is:
- write bit
- Branch based on newest dev
- CI green

### Versions

- Python:
- OS:
- Pymodbus:
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: tcp/rtu/ascii - sync/async
- Client: tcp/rtu/ascii - sync/async

### Description

### Code and Logs

```python
# code and logs here.

# please use the following to format logs when posting them here
import pymodbus

pymodbus.pymodbus_apply_logging_config()
```

janiversen, 2023-04-25:
@alexrudd2 can you please check, that you can merge a pull request now (I removed the demand for approval).

janiversen, 2023-04-25:
btw. if you are co-authoring a PR, you cannot approve it, that is standard GitHub security.

alexrudd2, 2023-04-25:
I don't have access to the Branch protection rules (which is fine), but I can guess as to the settings.  Note: the screenshots below are from another repo and don't show current values!

> write bit

Enabled already.

> Branch based on newest dev

I don't understand how, but I think this one works already?  Just to be explicit, you always use **Squash and merge**, right?  (The alternative is **rebase and merge**)
[image]

> CI green

Doesn't appear to be enabled
[image]

 > ~~Review approval~~

I believe you were previously able to bypass the requirement of approval because you're an administrator.
[image]

Anyways, it's disabled now.  https://github.com/pymodbus-dev/pymodbus/pull/1520 worked for me.

janiversen, 2023-04-25:
Please always use squash and merge. Your write bit allows you to merge PR, but not maintain the repo (settings).

I had to disable the CI green (or at least I think so) to ensure you can merge. However that is not a problem if we both are a bit careful.

alexrudd2, 2023-04-25:
> I had to disable the CI green (or at least I think so) to ensure you can merge. However that is not a problem if we both are a bit careful.

Hmm, I don't think this was necessary.  CI green should be a different status check than requiring review.

I think the below settings are what you want (although you'll have to change **test-python**)
[image]

closed 2023-04-25 by janiversen

janiversen, 2023-04-25:
Got it to work....you need a status check on top of "require branches...". It works now.