pymodbus-dev/pymodbus #1524: no readings from han port on power meter.

BrixSat, 2023-04-25:
This line as it is makes my power meter not be measured correctly.

https://github.com/pymodbus-dev/pymodbus/blob/c5772b35ae3f29d1947f3ab453d8d00df846459f/pymodbus/register_read_message.py#L47

I have to swith it to 1 +1 +2

janiversen, 2023-04-25:
You need some better arguments than that !

This line works in a bigger number of installations. And why do you want to change it to a fixed length, would it not be a lot easier to call with count=2 ( 1+ 1+ 2*2).

janiversen, 2023-04-25:
and just to clarify 2* self.count is because each register is defined to be 2 bytes, so to read 4 bytes you need count=2 in your call.

BrixSat, 2023-04-25:
Sorry sur, just to clarify

" return 1 + 1 + 2 * self.count "
should be 
" return 1 + 1 + 4 * self.count .

That was something i found out on the home assistant forum. And it does work in deed. At least for the HAN port of power meter devices,.

janiversen, 2023-04-26:
That only works if either you specify the wrong count, or your meter do not follow the standard modbus protocol, because 4* would mean each register is 4bytes.

BrixSat, 2023-04-26:
This is the protocol documentation I  have on the power meter. 

https://storage.googleapis.com/taikai-storage/images/challenges/hackelectron/DEF-C44-509N.PDF

[DEF-C44-509N.PDF](https://github.com/pymodbus-dev/pymodbus/files/11333180/DEF-C44-509N.PDF)

janiversen, 2023-04-26:
Ok, what do you expect me to do ?

Just to clarify, I am here to solve bugs in the software, to me it seems you want something which is non-standard  (I have had exactly the same discussion with another person both here and in homeassistant, long time ago).

If you want something non-standard you can either:
- fork and modify your own version
- submit a PR that allows non-standard without affecting the standard.

BrixSat, 2023-04-26:
Will do the second one eventually

closed 2023-04-27 by janiversen